ROBOTIS-GIT/turtlebot3 #704: SBC Setup should use nmcli

kydos, 2021-02-16:
First off my kudos for the wonderful work done with the turtlebots. That said, I think that the instruction on how to setup the network provided at https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup for foxy (and alike) should use **nmcli** to configure the WiFi -- perhaps as an alternative method. 

Using nmcli it would be as simple as 

```$ nmcli d wifi connect WIFI-SSID password <password>```

Of course this assuming that nmcli is installed -- which unfortunately is not on the image you provide. That could be solved by pre-installing it on your image or adding a reference to https://core.docs.ubuntu.com/en/stacks/network/network-manager/docs/installation

I'd be happy to contribute a pull request if that helps.

ROBOTIS-David, 2021-02-16:
@kydos 

Thank you for suggesting and contributing the nmcli. I will transfer this suggestion and my coworker will review the suggestion. 

Thank you :)

ROBOTIS-Will, 2021-02-16:
@kydos 
Thanks for the suggestion.
We're about to prepare the TurtleBot3 installation images for each ROS version.
This could be very useful after installing from the image then setting up the network.
Much appreciated!

kydos, 2021-03-15:
Sounds good, let me know if there is anything I can help with. Otherwise, I'll let you decide when you feel the issue should be closed.

sunghowoo, 2024-11-08:
This issue will be closed due to prolonged inactivity.

closed 2024-11-08 by sunghowoo