ROBOTIS-GIT/turtlebot3 #705: TB3-Manipulation manual links to a deprecated library

goekce, 2021-02-16:
https://github.com/ROBOTIS-GIT/emanual/blob/master/_includes/en/platform/turtlebot3/manipulation/manipulation_software_setup_dashing.md contains a link to the repo https://github.com/ROBOTIS-GIT/open_manipulator_with_tb3.git which seems to be deprecated.

Is the new repository https://github.com/ROBOTIS-GIT/turtlebot3_manipulation?

If the new repo is included in the manual there would be less confusion for the user.

ROBOTIS-Will, 2021-02-16:
Hi @goekce,
Sorry about the confusion and delayed response on this issue.
The `open_manipulator_with_tb3.git` contains ROS2 example of TurtleBot3 with OpenMANIPULATOR-X while `turtlebot3_manipulation` deals with ROS1 example.
Thank you.

goekce, 2021-02-16:
@ROBOTIS-Will does this mean that OpenMANIPULATOR-X will not be supported for ROS2?

ROBOTIS-Will, 2021-02-16:
@goekce 
The OpenMANIPULATOR currently supported in Dashing and will be updated to support the latest ROS2 including ROS1.
The exact schedule has not been confirmed yet, but I think it will begin in 2Q of this year.

goekce, 2021-02-17:
@ROBOTIS-Will good to hear that!

I understand that `open_manipulator_with_tb3.git` supports ROS2 Dashing and will be updated to support the latest ROS2 distro.

Why did `open_manipulator_with_tb3.git` then got archived?

ROBOTIS-Will, 2021-02-18:
@goekce 
The repository was once corrupted with improper branch merging in the past and I separated the usable commit for later work.
Most likely the update will be based on the archived source code.

goekce, 2021-02-18:
Thanks Will!

closed 2021-02-18 by goekce