ROBOTIS-GIT/turtlebot3 #708: Turtlebot with OM no movement in joint 1 (ID: 11)

VincentFokker, 2021-02-17:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ x] etc (Nvidia Jetson TX2)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ x] etc (Ubuntu 16.04 LTS)

5. Which OS you installed on Remote PC?

    - [ x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.2.4]
    - Firmware version: [2.0.1]

7. Specify the commands or instructions to reproduce the issue.

    - The device is setup with the firmware from the shell script upload method: https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation
    - When booting first time, there is no torque on the first joint in the chain of the OM (joint ID: 11). I am able to read the joint states through Rviz and the state of this joint is updated, however no control is possible. This problem only occurs with the tb3_with_om.opencr firmware. When using the usb_to_dxl and the manipulator controller, the joint does move as expected. I also noticed that with switching to protocol 1.0 the joint starts to operate, but movement is not fluid, so that is not ideal. Also, when the gripper is removed (in the Arduino script, or physically) the joint (ID: 11) starts operating normally.

8. Copy and Paste the error messages on terminal.

No Errors on terminal. When planning and executing paths with MoveIt, the controller gives a feedback that the execution was completed successfully.

9. Please describe the issue in detail.
The First joint of the OM arm on the turtlebot is not moving. Therefore the arm can not rotate. There is no torque on the motor after uploading the firmware. I have tried different motors (have two spare) but this did not solve the issue. As soon as an motor has the ID 11, it does not function any more. The problems seems to be in the firmware..

ROBOTIS-Will, 2021-02-18:
@VincentFokker 
Unfortunately, I wasn't able to reproduce the issue on my default hardware setup with Raspberry Pi 3B+.
Have you tried reading all connected DYNAMIXEL with [DYNAMIXEL Wizard 2.0](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/)?
If you haven't changed any DYNAMIXEL configuration, you should be able to find ID : `1`, `2`, `11`, `12`, `13`, `14`, `15` under the baudrate of `1Mbps`.

VincentFokker, 2021-02-18:
@ROBOTIS-Will 
Yes I did check all the motor settings with the Dynamixel Wizard 2.0. They are all set to baud 1MBps, with ids 1,2 for the wheels, and 11, 12, 13, 14 for the chain, 15 for the gripper. 
This hardware setup works fine with the USB to DXL, and then using the open manipulator controller running on the host PC. That is why my suspicion is on the firmware... Any other suggestions on how I can debug this?

ROBOTIS-Will, 2021-02-19:
@VincentFokker 
I confirm several times that the OpenCR firmware works fine with a default SBC(Raspberry Pi3B+).
The OpenCR firmware I tested was based on the eManual below.
https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#opencr-setup
Which power source do you use? Have you tried with the battery included in the TurtleBot3 package?

VincentFokker, 2021-03-03:
Hi, Yes we did try different power sources. Currently we sometimes use a 3S1P 11.1V 55.5Wh 5000mAh Lipo, and sometimes we use a direct power source from adapter (provided with the kit). We also tried with the provided battery included.

I Dont think power is the issue, as the full setup does work with other firmware (e.g. using USB to DXL and then controlling the manipulator arm from the PC).

Any other ideas or more suggestions on debugging?

ROBOTIS-Will, 2021-03-04:
@VincentFokker 
If you have a U2D2, could you connect it to any available port on OpenCR or DYNAMIXEL and get the whole list of packet transmitted and received?
As far as I tested on my TurtleBot3 setup with default hardware, I couldn't reproduce your issue and never heard of the similar symptom from other customers.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#packet

[image]

ROBOTIS-Will, 2021-03-08:
I'm going to close this thread as more information is available from the thread below.
https://forum.mybotshop.de/t/turtlebot-waffle-open-manipulator/206/17

closed 2021-03-08 by ROBOTIS-Will