ROBOTIS-GIT/turtlebot3 #709: TBS machine learning software setup error. Do we really use the tensor flow virtual env or not?

stevensu1838, 2021-02-20:
Hi team,

I followed your instructions on your e manual to install tensorflow and keras on my ROS kinetic PC. However
[e manual ](https://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#machine-learning)
[TurtleBot3 49 Machine Learning tutorial 1](https://www.youtube.com/watch?v=s0qgunKt654)

However, when I run import tensorflow as tf I get the following errors. I am setting up this course for my students. Could you please update your e manual and make it reliable? Thanks a lot. 

`mario@mario:~/Downloads$ python
Python 2.7.15 |Anaconda, Inc.| (default, May  1 2018, 23:32:55) 
[GCC 7.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorflow
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mario/anaconda2/lib/python2.7/site-packages/tensorflow/__init__.py", line 24, in <module>
    from tensorflow.python import pywrap_tensorflow  # pylint: disable=unused-import
  File "/home/mario/anaconda2/lib/python2.7/site-packages/tensorflow/python/__init__.py", line 47, in <module>
    import numpy as np
  File "/home/mario/anaconda2/lib/python2.7/site-packages/numpy/__init__.py", line 142, in <module>
    from . import core
  File "/home/mario/anaconda2/lib/python2.7/site-packages/numpy/core/__init__.py", line 91, in <module>
    raise ImportError(msg.format(path))
ImportError: Something is wrong with the numpy installation. While importing we detected an older version of numpy in ['/home/mario/anaconda2/lib/python2.7/site-packages/numpy']. One method of fixing this is to repeatedly uninstall numpy until none is found, then reinstall this version.
>>> import tensorflow as tf
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mario/anaconda2/lib/python2.7/site-packages/tensorflow/__init__.py", line 24, in <module>
    from tensorflow.python import pywrap_tensorflow  # pylint: disable=unused-import
  File "/home/mario/anaconda2/lib/python2.7/site-packages/tensorflow/python/__init__.py", line 47, in <module>
    import numpy as np
  File "/home/mario/anaconda2/lib/python2.7/site-packages/numpy/__init__.py", line 140, in <module>
    from . import _distributor_init
ImportError: cannot import name _distributor_init
>>> 
`

The last question is :
Why is it that u created a virtual environment called tensorflow but never activated it? Does it mean that ur tensorflow is installed in the root/base environment? In this case whats the use of creating a virtual env?

This question is asked on your youtube tutorial as well not answered yet. Cheers

stevensu1838, 2021-02-21:
Hi team,
I spend the whole to to try run your instructions on your e manual to install tensorflow and keras on my ROS kinetic PC. It didn't work. Would you please consider update your instruction to make it work? Thanks a lot. Cheers.

ROBOTIS-Will, 2021-03-10:
Hi @stevensu1838 
I'm sorry about the delayed response.
The TurtleBot3 machine learning package is being reviewed and will be updated soon for the latest software support.
Thank you.