ROBOTIS-GIT/turtlebot3 #710: Turtlebot3 Scan Not Working Properly

chuckl72, 2021-02-22:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.
[image]
[image]

 Roscore
roslaunch turtlebot3_bringup turtlebot3_robot.launch
export TURTLEBOT3_MODEL=burger
roslaunch turtlebot3_slam turtlebot3_slam.launch

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    The laser is not properly identifying any walls or objects.  See the first screen print which is after running for about 30 minutes.  Will, in another thread said "Looks like you are only receiving half of the scan topic as the frequency is 2.21Hz.
The frequency should be close to 5Hz (LDS sensor turns 5 times per second)."https://github.com/ROBOTIS-GIT/turtlebot3/issues/700#issuecomment-783080216  This value ranges from .38 to 2.5, and it doesn't matter if both the Turtlebot and PC are connected via ethernet or wifi.  Any ideas?

Chuck

[image]
[image]

ROBOTIS-Will, 2021-02-23:
Hi @chuckl72 
The bandwidth for each topic in the rqt is extremely low.
I'm not sure the exact reason of such low bandwidth, but have you performed time sync on both machines?
https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc

chuckl72, 2021-02-23:
Will

Yes I ran the time sync commands on both the pc and Turtlebot3.  I also connected both the pc and Turtlebot3 via ethernet cable which showed no improvement.  I want to look at the turtlebot3_slam/config/gmapping_params.yaml file to see if the update interval is ok, but I'm having problems finding the file.
Chuck

chuckl72, 2021-02-23:
I was able to locate the turtlebot3_slam/config/gmapping_params.yaml file and changed the map_update_interval to 10 from the setting of 2.  It didn’t make any difference.  Am I the only person that has these many problems?  Do I have a hardware problem?

ROBOTIS-Will, 2021-02-24:
Chuck,

Updating the interval will not work for your case since the communication between the robot and your PC is unstable.
The map update interval is only useful to reduce the burden of PC while robot should keep sending a data set regardless of its actual use.
Yours is a very unusual case that I don't have much information.
It could be your wifi router that creates a bottleneck.
Maybe trying with another router could help to identify the problem.

chuckl72, 2021-02-25:
Will,

I picked up another router - Linksys E1200, and set up a network without any internet connect.  The only two devices connected to the router were the Turtlebot3 and the PC.  Unfortunately, the results were the same, no green dots or walls identified.  Any other things we should try?

Chuck

[image]
[image]

ROBOTIS-Will, 2021-02-25:
@chuckl72 
In the RViz window, `Robot Model` and `LaserScan` are not available. Could you reload the SLAM node from your PC?
```bash
$ export TURTLEBOT3_MODEL=burger
$ roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping
```

After changing the router, the bandwidth of other topics looks good while LDS sensor related topics are still degraded.
[image]

This could mean the LDS sensor is not working properly.
I'd recommend to send the LDS module for a repair or evaluation (RMA).
Thank you.

maity-kgz, 2021-02-26:
Hi all,
I have the same issue with the LDS-01. The data from the Lidar is somehow discontinuous. Sometime I get full 360 degree data, some time I get only zeros or partially zero data. Very strange behaviour. Pluging, unplugging, restarting, closing all windows jalusi to eliminate the Sun did not help. Here is the short video of the issue https://youtu.be/lJQUFmwHxtM
Get the same issue with the lidar connected directly to the PC. Scan publish rate is about 2,5 Hz
TB3, burger, Ubuntu 16, Kinetic.

maity-kgz, 2021-02-26:
Got the same issue with Ubuntu 20, Noetic, Lidar connected directly to the PC.
[image]

chuckl72, 2021-02-26:
maity-kgz

I got a RMA yesterday from Robotis to return my LDS for evaluation, but based on you also having a problem, I'm going to hold off until Will has had a chance to comment.

Chuck

ROBOTIS-Will, 2021-03-02:
@maity-kgz 
Thank you for attaching the video
It looks like your Lidar does not work properly as I see a lot of zeros and inconsistent values.
Please make sure that all cables are undamaged and connectors are firmly attached to headers.
Also try cleaning up the flat face of the side of Lidar sensor with a soft cloth such as glass cleaning cloth.
If this doesn't help to resolve the issue, please request repair service for your sensor.

chuckl72, 2021-03-05:
Will,

It's working now after unplugging and replugging in the Lidar.  Thanks for the help.

Chuck

closed 2021-03-05 by chuckl72

maity-kgz, 2021-03-05:
In my case, plugin, unplugging, changing USB cables did not actually help.
There is one more symptom if I let the Lidar sit over a long time and test again, it is starting to working normally (5Hz, no zeros, etc), but only for few minutes. And eventually, fall back to the described state above.

ROBOTIS-Will, 2021-03-08:
@maity-kgz 
While your Lidar stalled, do you get any data updated in the rqt topic monitor?
What is it look like?