ROBOTIS-GIT/turtlebot3 #711: Pi loses wifi connection during setup

batuhanates, 2021-03-06:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    Follow e-Manual quick start guide for Melodic until the end of SBC setup then reboot Pi.

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    I was following the quick start guide and the wifi was working fine at first. After the OpenCR part I rebooted Pi and the wifi connection was gone. Couldn't find any solution and decided to start over. This time I rebooted Pi at the and of SBC Setup after doing USB Port setting `rosrun turtlebot3_bringup create_udev_rules` and again the wifi connection is gone. wlan0 doesn't show up if I check ifconfig.

ROBOTIS-Will, 2021-03-08:
@batuhanates 
Thank you for reporting the issue.
I'll prepare the recovery image so that you can try with it.
Please allow me a few days to deliver the test image.
Thank you.

ROBOTIS-Will, 2021-03-10:
@batuhanates 
Please refer to the thread below to test the Melodic image for Raspberry Pi 3B+
https://github.com/ROBOTIS-GIT/turtlebot3/issues/703#issuecomment-794854735
Thank you.

Nickname44, 2021-03-10:
I had the same problem. I solved it by installing the 18.04.5-preinstalled-server image instead of the 18.04.3 image, which can be found here: https://cdimage.ubuntu.com/releases/bionic/release/ubuntu-18.04.5-preinstalled-server-arm64+raspi3.img.xz

The rest of the quick start guide works fine with this image

ROBOTIS-Will, 2021-03-11:
@batuhanates 
I've tested the image multiple times before sharing and I just downloaded from the link and recovered successfully as below.
[image]

I used the `disks` tool in Ubuntu, or use Raspberry Pi imager.
FYI, the extracted image file size is about 5.2GB(5,242,880,000 bytes).

batuhanates, 2021-03-11:
@ROBOTIS-Will 
Thank you very much for your help! Seems like yesterday I messed up. Today I reinstalled with your image again and it works fine without any problems so far. So I'm closing the issue.
Thank you.

closed 2021-03-11 by batuhanates