ROBOTIS-GIT/turtlebot3 #712: Issues controlling OpenManipulator-X using Rviz or Robotics GUI Controller

Prajwalmj, 2021-03-11:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [v1.1.0]
    - Firmware version: [v2.0.1]

7. Specify the commands or instructions to reproduce the issue.

```bash
roslaunch turtlebot3_bringup turtlebot3_robot.launch     # warning message, [TurtleBot3 SBC]
roslaunch turtlebot3_manipulation_moveit_config move_group.launch     # error message
roslaunch turtlebot3_manipulation_moveit_config moveit_rviz.launch    # warning message
roslaunch turtlebot3_manipulation_gui turtlebot3_manipulation_gui.launch     # warning message
```

8. Copy and Paste the error messages on terminal.

    - Message from Rviz
    ```text
    process[rviz_larx_dell_3_12574_671864932024397051-1]: started with pid [12591]
    [ INFO] [1615494672.447645911]: rviz version 1.12.17
    [ INFO] [1615494672.447719651]: compiled against Qt version 5.5.1
    [ INFO] [1615494672.447741376]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1615494673.405339469]: Stereo is NOT SUPPORTED
    [ INFO] [1615494673.405402698]: OpenGl version: 4.6 (GLSL 4.6).
    [ INFO] [1615494677.058774560]: Loading robot model 'turtlebot3_manipulation'...
    [ INFO] [1615494677.178108958]: Loading robot model 'turtlebot3_manipulation'...
    [ INFO] [1615494677.251028366]: Using position only ik
    [ INFO] [1615494678.045440080]: Starting scene monitor
    [ INFO] [1615494678.050857070]: Listening to '/move_group/monitored_planning_scene'
[... 49 lines cut ...]
    [INFO] [1615494596.865602]: Setup subscriber on reset [std_msgs/Empty]
    [WARN] [1615494596.879623]: Failed to get param: timeout expired
    [INFO] [1615494596.883070]: Setup TF on Odometry [odom]
    [INFO] [1615494596.886641]: Setup TF on IMU [imu_link]
    [INFO] [1615494596.890209]: Setup TF on MagneticField [mag_link]
    [INFO] [1615494596.893646]: Setup TF on JointState [base_link]
    [INFO] [1615494596.901542]: --------------------------
    [INFO] [1615494596.905014]: Connected to OpenCR board!
    [INFO] [1615494596.908465]: This core(v2.0.1) is compatible with TB3 with OpenManipulator
    [INFO] [1615494596.911897]: --------------------------
    [INFO] [1615494596.915311]: Start Calibration of Gyro
    [INFO] [1615494596.918825]: Calibration End
    [ WARN] [1615494597.180745387]: This firmware(v2.0.1) isn't compatible with this software (v1.1.0)
    [ WARN] [1615494597.181006895]: You can find how to update its in `FAQ` section(turtlebot3.robotis.com)
    ```
[image]
[image]
[image]

9. Please describe the issue in detail.

    - Not able to control OpenManipulator-X interface with Waffle Pi (turtlebot3) either from Rviz or Robotics GUI Controller.
    - When using GUI Controller, Set/Open Gripper command manipulates other motors instead of the gripper. We have made sure that the order of Motor IDs assembled are correct.
    - Motors with ID 13 and 14 are movable by hands even we execute the controller while rest of the motors of the arm are locked.
    - This issue is similar to Issue#21 closed earlier. We tried the solution suggested there, but it did not solve the issue.

Prajwalmj, 2021-03-12:
- We assembled the arm according to the instructions in the official manual for OpenManipulator-X. We tested the working and control using Rviz, Robotics GUI controller. Motor with IDs 13, 14 could not be controlled by the commands from GUI controller and it could be moved by hand when it's turned ON, while other motors are locked. In addition to that, when we try to execute commands like "Set Gripper", base motor (ID - 11) moves instead of the gripper. Also, when we set joint angles and click send in the GUI controller, there is no response.
- We are currently using DYNAMIXEL XM430-W350-T.
- We follow the instructions from the following website: [assemble](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/assembly/#assembly-manual). The motor IDs are from 11-15. And we choose the motor IDs and order according to the assembly manual.
- We followed the instructions on [openmanipulator-x](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/#quick-start-guide) and [turtlebot3-manipulation](https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation) to download and install all the code from Github. We did it recently and we think it should be the latest version.

Please let us know if you need more information.
[image]

ROBOTIS-Will, 2021-03-15:
DYNAMIXEL only operate with a digital packet and correctly transmitted signal does not affect to other joints.
If ID 11 joint operates instead of the gripper, make sure to check all ID so that each joint is correctly assembled.
It looks like you have installed ID 15 to the base roll joint.
Also, there should be only one ID for each DYNAMIXEL and they should NOT be duplicated.

Prajwalmj, 2021-03-15:
- We have installed ID 11 only to the base joint, ID 12 for the 1st joint, ID 13 for the 2nd joint, ID 14 for the 3rd joint, ID 15 for the gripper as per the official manual. There are stickers on each DYNAMIXEL indicating and we have used them accordingly.
- We confirmed that all the IDs are different (based on stickers on DYNAMIXEL) and there aren't any duplicates.
- We have tested the gripper using GUI controller. We pressed different buttons many times, only a few times only motor ID 12 moves, rest of the time, nothing happens. We guess this is hardware issue. So we are planning to disassemble everything, replace jumper wires and test everything again.
- Could you suggest some other things that we could try? What about the firmware incompatibility issue and issues with Rviz?

Prajwalmj, 2021-03-26:
@ROBOTIS-Will , there were hardware issues with wiring and connections. After disassembling and assembling everything again, we figured out this. Everything is working fine now. Thanks for your replies and help! So, I am closing the the issue.

closed 2021-03-26 by Prajwalmj