ROBOTIS-GIT/turtlebot3 #714: Unable to launch empty world

ibrahimqazi, 2021-03-21:
Hello guys,
I want to run empty world file but it gives me an error. When I run this:
`roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch`

I get this error:
```
RLException: Invalid <arg> tag: environment variable 'TURTLEBOT3_MODEL' is not set. 

Arg xml is <arg default="$(env TURTLEBOT3_MODEL)" doc="model type [burger, waffle, waffle_pi]" name="model"/>
```

ROBOTIS-Will, 2021-03-23:
@ibrahimqazi 
Have you set the TURTLEBOT3_MODEL parameter as below?
https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#launch-simulation-world
```bash
$ export TURTLEBOT3_MODEL=burger
$ roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch
```

ibrahimqazi, 2021-03-23:
Thanks @ROBOTIS-Will  it worked

closed 2021-03-23 by ibrahimqazi