ROBOTIS-GIT/turtlebot3 #715: Failed to launch turtlebot telex using a launch file

vidurvij-Unity, 2021-03-22:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi
Trying to run it on Mac OS
2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

On my Mac OS

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [x ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    ros2 launch <path-to-the-launch-file

8. Copy and Paste the error messages on terminal.

    - INFO] [launch]: All log files can be found below /Users/vidur.vij/.ros/log/2021-03-22-03-27-28-647794-vidur.vij-0SZM-78566
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [teleop_keyboard-1]: process started with pid [78568]
[teleop_keyboard-1] Traceback (most recent call last):
[teleop_keyboard-1]   File "/Users/vidur.vij/dev_ws/install/turtlebot3_teleop/lib/turtlebot3_teleop/teleop_keyboard", line 33, in <module>
[teleop_keyboard-1]     sys.exit(load_entry_point('turtlebot3-teleop==2.1.1', 'console_scripts', 'teleop_keyboard')())
[teleop_keyboard-1]   File "/Users/vidur.vij/dev_ws/install/turtlebot3_teleop/lib/python3.8/site-packages/turtlebot3_teleop/script/teleop_keyboard.py", line 142, in main
[teleop_keyboard-1]     settings = termios.tcgetattr(sys.stdin)
[teleop_keyboard-1] termios.error: (102, 'Operation not supported on socket')
[ERROR] [teleop_keyboard-1]: process has died [pid 78568, exit code 1, cmd '/Users/vidur.vij/dev_ws/install/turtlebot3_teleop/lib/turtlebot3_teleop/teleop_keyboard --ros-args -r __node:=teleop_keyboard'].

9. Please describe the issue in detail.

   Modified the package to add a launch file in the package. The launch file can be found here. It produces the error pasted above.

`<launch>
  <set_env name="TURTLEBOT3_MODEL" value="waffle"/>  

  <node pkg="turtlebot3_teleop" exec="teleop_keyboard" name="teleop_keyboard"  output="screen">
  </node>
 
</launch>

`

ROBOTIS-Will, 2021-03-23:
@vidurvij-Unity 
I cannot find the modified launch file. 
Could you update your posting?

vidurvij-Unity, 2021-03-23:
> @vidurvij-Unity
> I cannot find the modified launch file.
> Could you update your posting?
<launch>
  <set_env name="TURTLEBOT3_MODEL" value="waffle"/>  

  <node pkg="turtlebot3_teleop" exec="teleop_keyboard" name="teleop_keyboard"  output="screen">
  </node>
 
</launch>

vidurvij-Unity, 2021-03-23:
<launch>
  <set_env name="TURTLEBOT3_MODEL" value="waffle"/>  

  <node pkg="turtlebot3_teleop" exec="teleop_keyboard" name="teleop_keyboard"  output="screen">
  </node>
 
</launch>

vidurvij-Unity, 2021-03-23:
`<launch>
  <set_env name="TURTLEBOT3_MODEL" value="waffle"/>  

  <node pkg="turtlebot3_teleop" exec="teleop_keyboard" name="teleop_keyboard"  output="screen">
  </node>
 
</launch>`

vidurvij-Unity, 2021-03-23:
```
<launch>
  <set_env name="TURTLEBOT3_MODEL" value="waffle"/>  

  <node pkg="turtlebot3_teleop" exec="teleop_keyboard" name="teleop_keyboard"  output="screen">
  </node>
 
</launch>

```

ROBOTIS-Will, 2021-03-24:
Please see below manual for creating the XML style launch file.
You should use the `param` tag in order to define a parameter.
https://design.ros2.org/articles/roslaunch_xml.html

vidurvij-Unity, 2021-03-29:
I don't get how this is the solution to the above problem. I need to set an environment variable not a parameter for the teleop_keyboard. Moreover python seems to be giving a socket error.

ROBOTIS-Will, 2021-03-30:
I think there are more useful resources from ROS Answers.
I'd recommend to search for a similar case you have.
Thanks.
https://answers.ros.org/questions/

sunghowoo, 2024-11-08:
This issue will be closed due to prolonged inactivity.

closed 2024-11-08 by sunghowoo