ROBOTIS-GIT/turtlebot3 #716: Turtlebot3 with OpenManipulator

johncabibihan, 2021-03-24:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - I would like to get the OpenManipulator working with Turtlebot 3 Waffle Pi

8. Copy and Paste the error messages on terminal.

    - There are no instructions for Noetic.

9. Please describe the issue in detail.

    - Please send the link for instructions. Thanks.

ROBOTIS-Will, 2021-03-24:
@johncabibihan 
The Manipulation section for the latest ROS is scheduled to be updated in 2Q of 2021.
Thank you.

MOJOliciousFTW, 2021-06-18:
@ROBOTIS-Will Q2 2021 is coming to an end. Any update? Would be awesome to try out TB3+OM on Melodic this summer.

ROBOTIS-Will, 2021-06-21:
Hi @MOJOliciousFTW 
Thanks for the interest.
I'm currently working on it, and having some hick ups but you'll be able to run the TB3 manipulation in this summer.
Thanks!

vale-max, 2022-05-11:
Hello @ROBOTIS-Will , is this project abandoned?
Can you provide at least some instructions to set it up with openmanipulator-x or anything that can make the manipulator usable?
Thanks

shiba-8ro, 2022-05-11:
@vale-max @ROBOTIS-Will 
Hi, 
OpenCR firmware for Noetic has not been released yet.
So, You shall use by modifying the sketch example of Arduino IDE.

・Step:
1. Arduno IDE, Examples> TurtleBot3> turtlebot3_with_open_manipulator> turtlebot3_with_open_manipulator_core open.
2. Uncomment the third line of turtlebot3_with_open_manipulator_core_config.h.
` #define NOETIC_SUPPORT       //uncomment this if writting code for ROS1 Noetic`
3. Write the sketch to OpenCR

sunghowoo, 2024-11-08:
This issue will be closed as the TB3+OM Noetic version has already been released.

closed 2024-11-08 by sunghowoo