ROBOTIS-GIT/turtlebot3 #717: Ros2 Foxy [turtlebot3_ros-3]: process has died [pid 4029, exit code -6, cmd '/home/odroid/tb3/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 --ros-args --params-file /home/odroid/tb3/install/turtlebot3_bringup/share/turtlebot3_bringup/param/burger.yaml'].

Fricodelco, 2021-03-24:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ x] etc Odroid-C2

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: latest foxy branch from git
    - Firmware version: [x.x.x]
    
7. Specify the commands or instructions to reproduce the issue.

    ros2 launch turtlebot3_bringup robot.launch.py

8. Copy and Paste the error messages on terminal.

[INFO] [launch]: All log files can be found below /home/odroid/.ros/log/2021-03-24-12-07-51-840364-odroid-4023
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
[INFO] [robot_state_publisher-1]: process started with pid [4025]
[INFO] [hlds_laser_publisher-2]: process started with pid [4027]
[INFO] [turtlebot3_ros-3]: process started with pid [4029]
[hlds_laser_publisher-2] [INFO] [1616587672.431440118] [hlds_laser_publisher]: Init hlds_laser_publisher Node Main
[hlds_laser_publisher-2] [INFO] [1616587672.431862149] [hlds_laser_publisher]: port : /dev/ttyUSB0 frame_id : base_scan
[robot_state_publisher-1] [WARN] [1616587672.436071462] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_link had 5 children
[robot_state_publisher-1] Link caster_back_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[robot_state_publisher-1] [INFO] [1616587672.450189510] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1616587672.450333521] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1616587672.450384525] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-1] [INFO] [1616587672.450422528] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-1] [INFO] [1616587672.450460530] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-1] [INFO] [1616587672.450495533] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-1] [INFO] [1616587672.450535536] [robot_state_publisher]: got segment wheel_right_link
[turtlebot3_ros-3] [INFO] [1616587672.465584654] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1616587672.467906826] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [INFO] [1616587672.468904901] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [1616587672.469892974] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[turtlebot3_ros-3] [ERROR] [1616587672.504347533] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-3] [INFO] [1616587672.506746712] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [1616587672.507061735] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [1616587672.507202745] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-3]   what():  could not create publisher: rcl node's context is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.10/src/rcl/node.c:441
[ERROR] [turtlebot3_ros-3]: process has died [pid 4029, exit code -6, cmd '/home/odroid/tb3/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 --ros-args --params-file /home/odroid/tb3/install/turtlebot3_bringup/share/turtlebot3_bringup/param/burger.yaml'].

9. Please describe the issue in detail.
Connection with Opener don't work
already done opener upgrade:
sudo dpkg --add-architecture armhf
sudo apt-get update
sudo apt-get install libc6:armhf
cd && rm -rf opencr_update.tar.bz2
wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2
tar -xjf ./opencr_update.tar.bz2
cd opencr_update
 ./update.sh /dev/ttyACM0 burger.opencr
and rules updated:
cd ~/turtlebot3_ws/src/turtlebot3/turtlebot3_bringup 
sudo cp ./99-turtlebot3-cdc.rules /etc/udev/rules.d/ 
sudo udevadm control --reload-rules 
sudo udevadm trigger

What else can I do?

Fricodelco, 2021-03-24:
Else when I tried to run just turtlebot3_node with this command:
$ ros2 run turtlebot3_node turtlebot3_ros 
I got this output:
[INFO] [1616588122.255842251] [turtlebot3_node]: Init TurtleBot3 Node Main
[INFO] [1616588122.256786298] [turtlebot3_node]: Init DynamixelSDKWrapper
[INFO] [1616588122.257610339] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[INFO] [1616588122.258499384] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[ERROR] [1616588122.292759097] [turtlebot3_node]: Failed connection with Devices
[INFO] [1616588122.294195168] [turtlebot3_node]: Add Motors
[INFO] [1616588122.294539186] [turtlebot3_node]: Add Wheels
[INFO] [1616588122.294693193] [turtlebot3_node]: Add Sensors
terminate called after throwing an instance of 'rclcpp::exceptions::ParameterNotDeclaredException'
  what():  sensors.bumper_1

ROBOTIS-Will, 2021-03-25:
hi @Fricodelco 
Thanks for your inquiry, but I'm afraid that we do not review issues for a customized product.
I'd recommend to visit [ROS Answers](https://answers.ros.org/questions/) for searching similar questions.
Thanks!

closed 2021-03-25 by ROBOTIS-Will