ROBOTIS-GIT/turtlebot3 #718: Cartographer installation issues

johncabibihan, 2021-03-26:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - I installed Cartographer by following the instructions at this link (other SLAM methods): https://emanual.robotis.com/docs/en/platform/turtlebot3/slam/#run-slam-node
    - 

8. Copy and Paste the error messages on terminal.

    - RuntimeError: Multiple packages found with the same name "cartographer":
- install_isolated/share/cartographer
- src/cartographer
[master] killing on exit

9. Please describe the issue in detail.

    - After installing cartographer, roscore does not work anymore. Please help. Thanks.

ROBOTIS-Will, 2021-03-29:
Hi,
Could you confirm how did you install ROS on Raspberry Pi?
The Raspbian image that we provide does not support ROS Noetic.
Installation instruction of cartographer is based on [the official Cartographer document](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html).
Try to uninstall the cartographer and install it again to see it fix the issue.
Thanks.

johncabibihan, 2021-03-29:
@ROBOTIS-Will : many thanks for the prompt help.

I uninstalled Cartographer. It did not return to the earlier working state with Gmapping. I decided to reinstall ROS on the PC. 

The bringup was successful but the SLAM (and Navigation) now have issues. There are errors in red. Your advice on how to resolve this will be very much appreciated. Thanks. 

[image]

[image]

ROBOTIS-Will, 2021-03-30:
Hi,
Please refer to below ROS answer thread.
Thanks.
https://answers.ros.org/question/313820/error-cannot-launch-node-of-type-gmappingslam_gmapping-cant-locate-node-slam_gmapping-in-package-gmapping/

johncabibihan, 2021-03-30:
@ROBOTIS-Will 

Many thanks for the link. Very useful. I was led to this link:

https://answers.ros.org/question/356105/slam-gmapping-for-ros-noetic/

This solved it:
$ sudo apt-get install ros-noetic-gmapping

sunghowoo, 2024-11-08:
This issue will be closed due to prolonged inactivity.

closed 2024-11-08 by sunghowoo