ROBOTIS-GIT/turtlebot3 #719: colcon build error when installing turtlebot3 packages on Master PC

Prajwalmj, 2021-03-27:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [x] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

```
      mkdir -p ~/turtlebot3_ws/src
      cd ~/turtlebot3_ws
      wget https://raw.githubusercontent.com/ROBOTIS-GIT/turtlebot3/ros2/turtlebot3.repos
      vcs import src < turtlebot3.repos
      colcon build --symlink-install
```
8. Copy and Paste the error messages on terminal.

```
larx@larx-lambda:~/turtlebot3_ws$ colcon build --symlink-install
Starting >>> turtlebot3_msgs
Starting >>> dynamixel_sdk
Starting >>> hls_lfcd_lds_driver
Starting >>> turtlebot3_description
Starting >>> turtlebot3_cartographer
Starting >>> turtlebot3_navigation2
Starting >>> turtlebot3_teleop
/home/larx/.local/lib/python3.6/site-packages/setuptools/dist.py:645: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
  % (opt, underscore_opt))
/home/larx/.local/lib/python3.6/site-packages/setuptools/dist.py:645: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  % (opt, underscore_opt))
Finished <<< turtlebot3_cartographer [0.44s]                              
Finished <<< turtlebot3_navigation2 [0.47s]                               
Finished <<< turtlebot3_description [0.50s]
Finished <<< hls_lfcd_lds_driver [0.51s]
Finished <<< dynamixel_sdk [0.53s]
--- stderr: turtlebot3_teleop                                              
/home/larx/.local/lib/python3.6/site-packages/setuptools/dist.py:645: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
  % (opt, underscore_opt))
/home/larx/.local/lib/python3.6/site-packages/setuptools/dist.py:645: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  % (opt, underscore_opt))
---
Finished <<< turtlebot3_teleop [0.62s]
Finished <<< turtlebot3_msgs [1.04s]                        
Starting >>> turtlebot3_node
Starting >>> turtlebot3_example
Starting >>> turtlebot3_fake_node
/home/larx/.local/lib/python3.6/site-packages/setuptools/dist.py:645: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
  % (opt, underscore_opt))
/home/larx/.local/lib/python3.6/site-packages/setuptools/dist.py:645: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  % (opt, underscore_opt))
Finished <<< turtlebot3_fake_node [0.50s]                                   
Finished <<< turtlebot3_node [0.52s]
Starting >>> turtlebot3_bringup
--- stderr: turtlebot3_example                                    
/home/larx/.local/lib/python3.6/site-packages/setuptools/dist.py:645: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
  % (opt, underscore_opt))
/home/larx/.local/lib/python3.6/site-packages/setuptools/dist.py:645: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  % (opt, underscore_opt))
---
Finished <<< turtlebot3_example [0.60s]
Finished <<< turtlebot3_bringup [0.20s]                    
Starting >>> turtlebot3
Finished <<< turtlebot3 [0.20s]                    
Starting >>> turtlebot3_gazebo
--- stderr: turtlebot3_gazebo                           
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Protobuf (missing: Protobuf_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindProtobuf.cmake:543 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:165 (find_package)
  CMakeLists.txt:23 (find_package)

---
Failed   <<< turtlebot3_gazebo [0.30s, exited with code 1]

Summary: 12 packages finished [2.42s]
  1 package failed: turtlebot3_gazebo
  3 packages had stderr output: turtlebot3_example turtlebot3_gazebo turtlebot3_teleop
  1 package not processed

```

9. Please describe the issue in detail.

    - I am trying to install ROS2 Dashing on turtlebot3
    - In Master PC, I have installed ROS2 [via debain packages](https://docs.ros.org/en/dashing/Installation/Linux-Install-Debians.html) as per the official [e-manual](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_setup/)
    - Then I installed ROS2 dependent packages as per the e-manual
    - When I try to install the turtlebot3 packages as per the e-manual, I get these errors
    - Also, after seeing the errors, I tried installing protobuf, but it was already installed. So, I could not figure out what is wrong.
    - Please let me know what is the issue, also if you need more information

ROBOTIS-David, 2021-03-30:
Hi, It seems that you are referencing the deprecated page. 

The page will be deleted sooner. 

Please reference the [Quick Start Guide](https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/) 
page. Make sure to select the right ROS version when getting in. 

I am wondering if you could tell me where were you linked from to the "[ros2_setup](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_setup/)" page? 

Note: The current e-manual explains :
"_Download the ubuntu-18.04.3-preinstalled-server-arm64+raspi3.img.xz image on your PC_", but not raspbian OS. 

Regards,

Prajwalmj, 2021-03-30:
@ROBOTIS-David 

- Thanks, I just googled Turtlebot3 ros2, got the [ros2_setup](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_setup/) page as the first suggestion.
- Yes, I also noticed that it's ubuntu server image file I need to install.
- Will try steps according to the link you sent and let you know the result.

ROBOTIS-David, 2021-03-31:
@Prajwalmj 

Thank you for letting me the deprecated pages can be reached by googling. 

I will fix it this week. Let me know if there is any other issues when following the current instruction. 

Regards,

Prajwalmj, 2021-04-01:
@ROBOTIS-David 

- I was able to successfully install ROS2 dashing on Master PC.
- having issues connecting to wifi network(actually it's mobile hot-spot connection for now) in turtlebot3 SBC. Followed steps according to [sbc_setup](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup) under Dashing tab
- I was able to connect first time, install all packages required without any issues. But, once I turned it off and tried to connect the next day, not able to connect. 
- So, I tried entering this command again

```
sudo netplan apply
```
and I get this warning.
```
Warning: The unit file, source configuration file or drop-ins of netplan-wpa-wlan0.service changed on disk. Run 'systemctl daemon-reload' to reload units. 
```
- Even after running that command, nothing changed, still could not connect to the network.
- I tried searching for this issues, and tried looking under folder /sys/class/net and found out there is only 2 folders, eth0, l0, no folder starting from w.
- I just want to what can I do to connect properly again. I checked the 50-cloud-init.yaml file again to see if I have edited it wrong, but since i was able successfully connect the first time, everything looks good, checked once again just to confirm.

ROBOTIS-Will, 2021-04-01:
hi @Prajwalmj 
Similar issue had been reported in Melodic which uses Ubuntu 18.04 as well.
This might be useful to update the wifi driver to activate the wlan0.
https://community.robotsource.org/t/sbc-wifi-problem-with-melodic/1995/9?u=robotis-will
Just in case you might not have ethernet, please use files attached in this thread.
- [brcmfmac43455-sdio.zip](https://github.com/ROBOTIS-GIT/turtlebot3/files/6246937/brcmfmac43455-sdio.zip)

Prajwalmj, 2021-04-14:
@ROBOTIS-Will , sorry about the late reply and thanks for your suggestion, it's working now! So I am closing this issue

closed 2021-04-14 by Prajwalmj