ROBOTIS-GIT/turtlebot3 #720: RobotModel is not present in tb3_navigation2.rviz

ahtsan, 2021-03-30:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [x] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

7. Specify the commands or instructions to reproduce the issue.

    - ros2 run rviz2 rviz2, and select the rviz file from https://github.com/ROBOTIS-GIT/turtlebot3/blob/ros2/turtlebot3_navigation2/rviz/tb3_navigation2.rviz

9. Please describe the issue in detail.

    - The Robot Model is missing, not consistent with the behavior with the rviz in ROS1 (https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_navigation/rviz/turtlebot3_navigation.rviz#L57)

ROBOTIS-Will, 2021-04-01:
Hi @ahtsan 
TurtleBot3 now uses [a default RViz view setting](https://github.com/ros-planning/navigation2/tree/main/nav2_bringup/bringup/rviz) from `nav2_bringup` package when launching Navigation.
The Nav2 RViz setting contains the Robot Model, but disabled by default.
If you need to see the robot model, please add below code underneath the `Displays:` in `Visualization Manager:` section of your `.rviz` file.
Thank you.
```
    - Alpha: 1
      Class: rviz_default_plugins/RobotModel
      Collision Enabled: false
      Description File: ""
      Description Source: Topic
      Description Topic: /robot_description
      Enabled: true
      Links:
        All Links Enabled: true
        Expand Joint Details: false
        Expand Link Details: false
        Expand Tree: false
        Link Tree Style: Links in Alphabetic Order
        base_footprint:
          Alpha: 1
          Show Axes: false
          Show Trail: false
        base_link:
          Alpha: 1
          Show Axes: false
          Show Trail: false
          Value: true
        base_scan:
          Alpha: 1
          Show Axes: false
          Show Trail: false
          Value: true
        caster_back_link:
          Alpha: 1
          Show Axes: false
          Show Trail: false
          Value: true
        imu_link:
          Alpha: 1
          Show Axes: false
          Show Trail: false
        wheel_left_link:
          Alpha: 1
          Show Axes: false
          Show Trail: false
          Value: true
        wheel_right_link:
          Alpha: 1
          Show Axes: false
          Show Trail: false
          Value: true
      Name: RobotModel
      TF Prefix: ""
      Unreliable: false
      Update Interval: 0
      Value: true
      Visual Enabled: true
```

ahtsan, 2021-04-01:
Thanks! Do you see any value in enabling the robot model by default in turtlebot3_navigation2? I would prefer seeing the robot by default as a user.

ROBOTIS-Will, 2021-04-01:
In case of TurtleBot3, the robot model in RViz doesn't really necessary as other TF information will provide enough information.
However, if you are operating a robot with multi DOF joints(such as manipulator), you might want to see the shape of the robot on RViz.

ahtsan, 2021-04-02:
Got it. For the sake of consistency (since the robot model is present in the rviz in ROS1), I still think there's some value there. I'll resolve it for now and come up with a pull request later if possible.

closed 2021-04-02 by ahtsan

ROBOTIS-Will, 2021-04-05:
@ahtsan 
Thanks for the feedback. I'll update the rviz configuration file in the next update so that developers can modify for their own purpose.

ahtsan, 2021-04-16:
@ROBOTIS-Will Great, thanks!