ROBOTIS-GIT/turtlebot3 #721: colcon build error when installing turtlebot3_node packages on ubuntu18-eloquent

miku54, 2021-04-01:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [✔ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [✔ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [✔ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [✔ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [✔ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - 1、Load the source code offline on and off github
    - 2、     
    -  mkdir -p ~/turtlebot3_ws/src
      cd ~/turtlebot3_ws
      colcon build --packages-select turtlebot3_node 

8. Copy and Paste the error messages on terminal.

    - HERE
Starting >>> turtlebot3_node
--- stderr: turtlebot3_node                             
libturtlebot3_node_lib.a(sensor_state.cpp.o): In function `rclcpp::Publisher<turtlebot3_msgs::msg::SensorState_<std::allocator<void> >, std::allocator<void> >::Publisher(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&)':
sensor_state.cpp:(.text._ZN6rclcpp9PublisherIN15turtlebot3_msgs3msg12SensorState_ISaIvEEES4_EC2EPNS_15node_interfaces17NodeBaseInterfaceERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_3QoSERKNS_29PublisherOptionsWithAllocatorIS4_EE[_ZN6rclcpp9PublisherIN15turtlebot3_msgs3msg12SensorState_ISaIvEEES4_EC5EPNS_15node_interfaces17NodeBaseInterfaceERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_3QoSERKNS_29PublisherOptionsWithAllocatorIS4_EE]+0x3c): undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<turtlebot3_msgs::msg::SensorState_<std::allocator<void> > >()'
libturtlebot3_node_lib.a(sound.cpp.o): In function `rclcpp::Service<turtlebot3_msgs::srv::Sound>::Service(std::shared_ptr<rcl_node_t>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::AnyServiceCallback<turtlebot3_msgs::srv::Sound>, rcl_service_options_t&)':
sound.cpp:(.text._ZN6rclcpp7ServiceIN15turtlebot3_msgs3srv5SoundEEC2ESt10shared_ptrI10rcl_node_tERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_18AnyServiceCallbackIS3_EER21rcl_service_options_t[_ZN6rclcpp7ServiceIN15turtlebot3_msgs3srv5SoundEEC5ESt10shared_ptrI10rcl_node_tERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_18AnyServiceCallbackIS3_EER21rcl_service_options_t]+0x7c): undefined reference to `rosidl_service_type_support_t const* rosidl_typesupport_cpp::get_service_type_support_handle<turtlebot3_msgs::srv::Sound>()'
collect2: error: ld returned 1 exit status
make[2]: *** [turtlebot3_ros] Error 1
make[1]: *** [CMakeFiles/turtlebot3_ros.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< turtlebot3_node [5.16s, exited with code 2]

Summary: 0 packages finished [6.96s]
  1 package failed: turtlebot3_node
  1 package had stderr output: turtlebot3_node
9. Please describe the issue in detail.

    - HERE
My SBC(Single Board Computer) is jetson nano ;
In jetson nano, I have successfully installed ROS-eloquent,. I tried to run the routine of navigation2, but compiled it incorrectly.
Please let me know what is the issue, also if you need more information

ROBOTIS-Will, 2021-04-01:
hi @miku54 
Please understand that we do not debug issues for the robot that is modified from the original component.
Since ROS2 Eloquent is deprecated, it is recommended to use other versions.
Thanks.

akshay-30, 2021-07-30:
Facing similar issues for ros2 foxy e

ROBOTIS-Will, 2021-08-05:
Hi @akshay-30 
I've just cloned the latest code and built them on my Foxy machine, but wasn't able to find an error during build.
Could you submit a new issue ticket for your case and attach the build log?
Thanks

sunghowoo, 2024-11-08:
This issue will be closed due to prolonged inactivity.

closed 2024-11-08 by sunghowoo