ROBOTIS-GIT/turtlebot3 #722: RTAB-map simulation in gazebo for turtlebot3

KenaHemnani, 2021-04-02:
Hii, I wanted to have suggessions regarding the following:    how shall I implement rtab-map in ROS i.e, simulation using gazebo for turtlebot3. If there are relevant resources please let me know. I have seen the official roswiki page for rtab-map but that shows mapping pacakge for turtlebot and not turtlebot3??
ISSUE TEMPLATE ver. 0.4.0

I am using ROS melodic.

ROBOTIS-Will, 2021-04-07:
Hi @KenaHemnani 
Unfortunately, we do not have an example for rtabmap.
You may find more useful information from the TurtleBot3 section of the wiki page below.
http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot
Thanks.

sunghowoo, 2024-11-08:
This issue will be closed due to prolonged inactivity.

closed 2024-11-08 by sunghowoo