ROBOTIS-GIT/turtlebot3 #723: `turtlebot3_cartographer` does not work in a TB3 simulation

goekce, 2021-04-05:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

- Waffle Pi

2. Which ROS is working with TurtleBot3?

    - ROS 2 Foxy Fitzroy

5. Which OS you installed on Remote PC?

    - Archlinux

7. Specify the commands or instructions to reproduce the issue.

```sh
ros2 launch turtlebot3_gazebo turtlebot3_house.launch.py
ros2 launch turtlebot3_cartographer cartographer.launch.py
```

8. Copy and Paste the error messages on terminal.

```
...
[rviz2-3] [ERROR] [1617634654.223115870] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
[rviz2-3] active samplers with a different type refer to the same texture image unit
[rviz2-3] [INFO] [1617634656.017648141] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 247.401 for reason 'Unknown'
[rviz2-3] [INFO] [1617634656.209632296] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 247.601 for reason 'Unknown'
... last message repeats
```

9. Please describe the issue in detail.

Using `turtlebot` v. 2.1.1 and `turtlebot_simulations` v. 2.2.2 which I built myself. I want to use cartographer in a simulated world, but `tf` seems to have problems as seen here:

[image]

Cases tested:

- `rviz` without cartographer (`ros2 launch turtlebot3_bringup rviz2.launch.py`): `tf` works.
- cartographer with real TB3 without simulation: `tf` works.

The problem could be that the `map` frame is not sent properly (maybe sent [without timestamp](https://answers.ros.org/question/357762/slam_toolbox-message-filter-dropping-message/?answer=357849#post-id-357849))

Can someone help me?

ROBOTIS-Will, 2021-04-06:
hi @goekce 
When running a simulation, please append `use_sim_time:=True` option as below.
```bash
$ ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=True
```

goekce, 2021-04-07:
Will, `use_sim_time` option solves the problem, thanks! There are still two issues:

1) when I restart the simulation using `Ctrl+r`, the message `[rviz2]: Message Filter dropping message: frame 'base_scan' at time` comes back and repeats again and the robot vanishes from rviz. Do I have to pass `use_sim_time` at restart again?

2) in case when everything works, I see that the frames `odom` and `map` seem the be skewed. Does `odom` start with a non-zero value? This issue may not be related to cartographer at all. Can you confirm this?

goekce, 2021-04-07:
After having read [ROS design document about clock and time](http://design.ros2.org/articles/clock_and_time.html) I suppose that `cartographer` may not support going back in time. That may be the reason why TF messages about `base_scan` are simply filtered as non-sense. What do you think?

ROBOTIS-Will, 2021-04-07:
Resetting the Gazebo simulation will also resets the `sim time` which will cause time difference between Gazebo and Rviz.
After resetting the gazebo, you should relaunch the rviz.

Odom is published by OpenCR in TurtleBot3. This will not be reset until TurtleBot3 is reset (or OpenCR is reset).

goekce, 2021-04-07:
> After resetting the gazebo, you should relaunch the rviz
:+1: 

> Odom is published by OpenCR in TurtleBot3. This will not be reset until TurtleBot3 is reset (or OpenCR is reset).

This makes sense. But when I start a new fake node for simulation, then odom should be zero as the model does not store any values like OpenCR.

It seem there is an odom offset preprogrammed in Turtlebot3 simulation model to set the zero point at the center of the Turtlebot3 world. To observe that:

```sh
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=True
```
you will see that `odom` is in the center of the TB world. Then close all windows and start:

```sh
ros2 launch turtlebot3_gazebo turtlebot3_house.launch.py
ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=True
```
The same offset can be observed. Can someone verify this?

ROBOTIS-Will, 2021-04-08:
You might want to checkout below document regarding the `map` and `odom` frames
https://www.ros.org/reps/rep-0105.html#coordinate-frames

closed 2021-04-08 by goekce