ROBOTIS-GIT/turtlebot3 #728: Robot Stopped Communicating after a Few Moments in SLAM or Tele Modes

LW-G38, 2021-04-12:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.15.9]
    - Firmware version: [1.2.6]

7. Specify the commands or instructions to reproduce the issue.

    - roslaunch turtlebot3_slam turtlebot3_slam.launch
    - roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

8. Copy and Paste the error messages on terminal.

    - No error message were shown.

9. Please describe the issue in detail.

    - When entering SLAM mode using teleop (https://emanual.robotis.com/docs/en/platform/turtlebot3/slam/#run-slam-node) or use teleop alone, the robot moves okay at the beginning. But after a few moments, the robot will stop moving without having any obstacles nearby, and when I check the dynamixel motors, I find them locked and the wheel cannot spin anymore. There is no battery shortage. Not sure if there are any operational mistakes from me, or it is a bug.
    - There is no updated topic values coming from the robot. I found out that this situations happens not only under SLAM teleop mode, but also normal teleop mode.
    - When this happens in SLAM mode, I killed the SLAM process and relaunch it, and find the map is empty and also I cannot make the robot move unless I reboot the robot.

ROBOTIS-Will, 2021-04-13:
Hi,
It is difficult for me to identify what causes your system stall.
You might find more useful information from Raspberry Pi forum.
Make sure that your RPi4 is still connected to the network when the symptom occurs.
I'd also check the system log and see what is happening.
Faulty SD card may also causes unexpected issues.

LW-G38, 2021-04-15:
Thank you Will. I'm thinking about what I could have done wrong, there are some of my guesses:

1. The voltage of Raspberry Pi may be unstable? I use the battery in the package, I'm not sure if these similar situations happened before.
2. I use a virtual machine from my Mac as the remote PC. Does this cause problem?

Thanks,

Wei

ROBOTIS-Will, 2021-04-15:
Hi @LW-G38 
The battery that comes with TurtleBot3 can provide enough power to run RPi4, but if you are heavily using RPi resource, you might want to add heat sink or a fan for active cooling.
Virtual machine is usually not recommended as there could be various issues. Issues occurred when using a virtual machine is difficult to debug.
Thank you.

LW-G38, 2021-04-16:
Thanks Will. Is there any place I can check the log file of the turtlebot? Or should I use a monitor to connect to the Pi or use VNC to observe?

LW-G38, 2021-04-16:
I may need to re-install the operating system of my Pi. In this case, do I need to set up OpenCR from scratch as well? Thank you.

ROBOTIS-Will, 2021-04-19:
Please find the Ubuntu system log for your Raspberry Pi and see if there's any issue with your system.
At this point I cannot assure that your issue is directly related to TurtleBot3 packages.

You don't need to setup the OpenCR again as you have already uploaded the Noetic firmware to OpenCR.
Thank you.

nullbyte91, 2021-08-20:
I'm also facing the same issue. When I use the Lithium-ion battery as a power supply, the motor stuck after a few moments. However, the system working completely fine with a direct 12V power supply.

ROBOTIS-Will, 2021-08-23:
@nullbyte91 
Your case is related to the power source. Make sure to use the official TurtleBot3 3S Li-Po battery, and check if the battery is in a good condition.
If the battery had been fully depleted before, it may have a damaged cell and may not fully functional due to the damaged cell.

nullbyte91, 2021-08-23:
@ROBOTIS-Will I got the waffle pi a week before, and I fully charged my battery before I use.

ROBOTIS-Will, 2021-08-30:
@nullbyte91 
Sorry about the delayed response.

> the motor stuck after a few moments

Does this mean the wheels are locked up with torque on? or can you rotate them by hand?

What voltage do you read from the battery status message?
[image]

sunghowoo, 2024-11-08:
This issue will be closed due to prolonged inactivity.

closed 2024-11-08 by sunghowoo