ROBOTIS-GIT/turtlebot3 #732: Problem with gazebo simulation.

DeepIndigo9, 2021-04-16:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [A] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [A ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ A] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [A ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - $ export TURTLEBOT3_MODEL=burger
    - roslaunch turtlebot3_gazebo turtlebot3_world.launch
  
8. Copy and Paste the error messages on terminal.

    - RLException: [turtlebot_world.launch] is neither a launch file in package [turtlebot3_gazebo] nor is [turtlebot3_gazebo] a launch file name  The tracetrack for the excepction was written to the log file.

9. Please describe the issue in detail.

    - I was trying to run a gazebo simulation detailed in the emanual of the Turtlebot3 page, but everytime i tried to run it the same error appeared.

ROBOTIS-Will, 2021-04-19:
Hi @DeepIndigo9 

You might have not setup the catkin workspace correctly.
The command below will append the workspace setup to `.bashrc` file and will be automatically loaded whenever you open a new terminal.
```
$ echo "source /opt/ros/noetic/setup.bash" >> ~/.bashrc
$ echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
$ source ~/.bashrc
```

Vaibbhav22134, 2023-08-26:
working

closed 2024-11-08 by sunghowoo