ROBOTIS-GIT/turtlebot3 #734: TB3 Waffle Pi locks up

BManCan, 2021-04-21:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [x ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
    - Could not find software and firmware version in Bringup messages

7. Specify the commands or instructions to reproduce the issue.

    - $ ssh ubuntu@192.168.2.240
    - $ ros2 launch turtlebot3_bringup robot.launch.py

8. Copy and Paste the error messages on terminal.

    - No error message

9. Please describe the issue in detail.

    - When I launch the bringup for waffle Pi through SSH, everything starts communicating normal. The orange status lights on OpenCR are flashing and I am able to start teleop by keyboard or using the RC100 remote. This will work briefly, then suddenly the orange lights stop flashing and the TB3 motors stop, Lidar keeps spinning. The Bringup terminal through SSH is also frozen and cannot be stopped using Ctl-C. Typically I have to reboot TB3 in order to reconnect through SSH again.

ROBOTIS-Will, 2021-04-22:
Hi @BManCan ,

It looks like your RPi is lost connection with the Remote PC or stalled for some reason.
If you can access your wifi router, please check if the Raspberry Pi is still connected to the router.
Defected SD card may also cause unexpected behavior of RPi.
Thank you.

BManCan, 2021-04-22:
It appears my problem is actually the same as in issue 546.
https://github.com/ROBOTIS-GIT/turtlebot3/issues/546
I am going to try Ubuntu 18.04.3 as suggested.

ROBOTIS-Will, 2021-04-22:
FYI, Dashing will reach to its EOL next month.
If I may ask, do you have any specific reason to use Dashing for your TurtleBot3?
There will be a major update for ROS2 Foxy, but you can still use SLAM and Navigation with your TurtleBot3 Waffle Pi.

BManCan, 2021-04-22:
I am new to the TurtleBot3 and wasn’t sure which version to go with.  I noticed that Dashing had more support for the Examples in the e-Manual.
I did not realize that camera support was only currently available on Kinetic.
Anyway I am slowly learning and working through the startup and installation. Do you recommend ROS2 Foxy as the preferred installation?
Thanks.

ROBOTIS-Will, 2021-04-22:
Foxy is the most stable version among currently available ROS2, and it is strongly recommended unless you are running ROS1.
We're working on a substitution for the deprecated [raspicam_node](https://github.com/UbiquityRobotics/raspicam_node) so camera will soon be avaialble.
Thank you.

BManCan, 2021-04-23:
Ok I installed 18.04.3 as suggested in
https://github.com/ROBOTIS-GIT/turtlebot3/issues/546

then resolved the missing wifi that happens after apt update && upgrade by restoring wifi firmware
https://community.robotsource.org/t/sbc-wifi-problem-with-melodic/1995/9

The SBC lockup issues are resolved.

closed 2021-04-23 by BManCan