ROBOTIS-GIT/turtlebot3 #736: turtlebot3_node colcon build error ros2 rolling

MoritzWillmann, 2021-04-22:
1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [x] ROS 2 Rolling (built from source)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

```bash
mkdir -p ~/tb3_rolling/src && cd tb3_rolling/src
wget https://raw.githubusercontent.com/ROBOTIS-GIT/turtlebot3/ros2/turtlebot3.repos
vcs import src < turtlebot3.repos
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro rolling
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```

8. Copy and Paste the error messages on terminal.

```bash
/home/moritz/ros2_rolling/install/rclcpp/include/rclcpp/node_impl.hpp: In instantiation of ‘auto rclcpp::Node::declare_parameter(const string&, const ParameterDescriptor&, bool) [with ParameterT = unsigned int; std::string = std::__cxx11::basic_string<char>; rcl_interfaces::msg::ParameterDescriptor = rcl_interfaces::msg::ParameterDescriptor_<std::allocator<void> >]’:
/home/moritz/tb3_rolling/src/turtlebot3/turtlebot3/turtlebot3_node/src/turtlebot3.cpp:57:55:   required from here
/home/moritz/ros2_rolling/install/rclcpp/include/rclcpp/node_impl.hpp:223:26: error: call of overloaded ‘ParameterValue(<brace-enclosed initializer list>)’ is ambiguous
  223 |   rclcpp::ParameterValue value{ParameterT{}};
```

9. Please describe the issue in detail.

I'm currently trying to setup the turtlebot3 packages on my host pc in combination with ros2 rolling. Whilst building the turtlebot3 packages as stated above I get the mentioned error message. I checked that only rolling is sourced for both ros2 and navigation2.

ROBOTIS-Will, 2021-04-22:
@MoritzWillmann 
Thank you for reporting the issue. I'm currently working on to release rolling packages, and for `turtlebot3` package, you may use the `galactic-devel` branch for now as I'm going to create the rolling branch very soon.
Thanks!

abhishek47kashyap, 2021-06-13:
Had the same problem.

Making the changes in `turtlebot3_node/src/turtlebot3.cpp` that @m2-farzan did seems to have resolved it and the build proceeded to completion (with a couple of warnings). But then at the end there's `1 package had stderr output: turtlebot3_node`.

Strange because there were no errors and there's `Finished <<< turtlebot3_node [1min 30s]` alright

m2-farzan, 2021-06-13:
'stderr' output is caused by errors and/or warnings. In your case it's probably just the warnings.

closed 2024-12-05 by sunghowoo