ROBOTIS-GIT/turtlebot3 #737: Turtlebot3 with Open Manipulator X Navigation Errors

slavenpetkovic, 2021-04-25:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.2.5]
    - Firmware version: [2.0.1]

7. Specify the commands or instructions to reproduce the issue.

    - Tried to follow Turtlebot3 emanual - Navigation section and when i run turtlebot3_navigation.launch file i get this (image below).

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - I run SLAM launch file (gmapping used) and i saved that map but when I try to run navigation i get these errors.. How to solve them?
    - And second question : "export TURTLEBOT3_MODEL=model"
    If I use turtlebot3 (waffle pi) with open manipulator x, which model do I need to provide here (waffle_pi or something else) and is it necessary to do every time or just once?
[image]

ROBOTIS-Will, 2021-04-26:
Hi @slavenpetkovic 
Thank you for reporting the issue.

If you haven't properly cofigured your TURTLEBOT3_MODEL parameter, this could happen. Please use the below option when using OpenMANIPULATOR-X on top of TurtleBot3 Waffle Pi.
```bash
$ export TURTLEBOT3_MODEL=waffle_pi
```
If you want to skip this command in the future, you can add it into your `.bashrc` as below.
```bash
$ echo 'export TURTLEBOT3_MODEL=waffle_pi' >> ~/.bashrc
$ source ~/.bashrc
```
Next time when you open your terminal, TURTLEBOT3_MODEL will be automatically set to `waffle_pi`.

When getting those errors in your RViz, make sure to perform the [2D Pose Estimate](https://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#estimate-initial-pose) and see if it helps to resolve the issue.

If Remote PC and TurtleBot3 SBC is not time synced, ROS messages may not properly accepted. Please try [Time Sync](https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc) on both Remote PC and TurtleBot3 SBC

Thank you.

slavenpetkovic, 2021-04-26:
@ROBOTIS-Will 
Thank you for your answer, it seems like Time Sync was the problem.
I have another question, is it possible to have Ros Kinetic version on Turtlebot3 with OP-X and Melodic ROS version on Ros Master Computer?
We made some simulations and other stuff for ROS Melodic and it would be hard to move all that to ROS Kinetic on PC.
Or is there a way to make Turtlebot3 with OP-X work on ROS Melodic? I asked for that tutorial in another ISSUE and you told me that NOETIC tutorital for TB3 with OPX will be published in Q2 2021. You didn't tell anything about Melodic.

ROBOTIS-Will, 2021-04-27:
Hi @slavenpetkovic 
Although ROS Melodic and Kinetic are not much different in terms of messages they use, it is recommended to run on a same ROS version.
I've run SLAM and Navigation on Melodic + Kinetic TB3 and vice versa, but this doesn't guarantee that your application would work flawlessly.
The Noetic is that last ROS1 version and has the [longest lifespan among ROS1](http://wiki.ros.org/Distributions), so Noetic will be the first one to be updated, but during the update, we'll keep looking for a chance to support Melodic as well.
Thank you.

slavenpetkovic, 2021-04-27:
Thanks @ROBOTIS-Will 
I have one more question about navigation. How to adjust the parameters so that the robot comes as close as possible to the obstacle? I followed Navigation section tutorial for setting parameters and tried different values and different combinations of inflation_radius and cost_scaling_factor but it simply won't get to about 2-3cm from obstacle like i want it to do. It stays at 20-25 cm and when I try to send him command to go closer, it starts spinning and goind backwards.
Do I need to change something when creating MAP using SLAM and RVIZ, or do I need to come as close as possible to obstacle when creating map or is it something else?

ROBOTIS-Will, 2021-04-28:
@slavenpetkovic 
You may change the [costmap_common_params_waffle_pi.yaml](https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_navigation/param/costmap_common_params_waffle_pi.yaml) as below.
The inflation_radius and the cost_scaling_factor will correctly reflected in the RViz.
```
obstacle_range: 3.0
raytrace_range: 3.5

footprint: [[-0.205, -0.155], [-0.205, 0.155], [0.077, 0.155], [0.077, -0.155]]
#robot_radius: 0.17

inflation_layer:
  enabled:              true
  inflation_radius:     0.55
  cost_scaling_factor:  10.0

map_type: costmap
observation_sources: scan
scan: {sensor_frame: base_scan, data_type: LaserScan, topic: scan, marking: true, clearing: true}
```

I'd recommend use a different approach to get close to an object instead of implementing it through navigation stack.
Navigation is focused on moving the platform with minimum collision from obstacles.
Thanks!

ROBOTIS-Will, 2021-06-10:
Closing this thread as there isn't recent activity.

closed 2021-06-10 by ROBOTIS-Will