ROBOTIS-GIT/turtlebot3 #738: Turtlebot3 - Burger Error 'cdc_acm 1-1.2:1.0: failed to set dtr/rts'

junu5121, 2021-05-03:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ v] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [v ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [v ] etc (Raspberry Pi 3B)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [v] etc (ubuntu-18.04.5-desktop-amd64)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [v ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - $ cd ./opencr_update
    - $ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr

8. Copy and Paste the error messages on terminal.

    - cdc_acm 1-1.2:1.0: failed to set dtr/rts

9. Please describe the issue in detail.

    - I followed this link(https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup).
    - In 'Step 3.3 OpenCR Setup', I proceeded as shown in Step. 5.
    - Then, the error message appears.
    - I entered the recovery mode as shown in Step. 7. 
    - And I repeat the Step. 1~5. But It doesn't work. Same error appears.
  
[image]

ROBOTIS-Will, 2021-05-04:
Hi @junu5121 
Have you set up the USB Port in the SBC section as below?
```bash
$ rosrun turtlebot3_bringup create_udev_rules
```

junu5121, 2021-05-05:
> Hi @junu5121
> Have you set up the USB Port in the SBC section as below?
> 
> ```shell
> $ rosrun turtlebot3_bringup create_udev_rules
> ```
> 
> Once you see the `[OK] jump_to_fw`, the firmware is successfully uploaded to OpenCR.
> I think there's something wrong with your serial port setting with Linux.

Yes, I did. But It didn't work....

ROBOTIS-Will, 2021-06-10:
@junu5121 
Sorry about the delayed response.
Have you tried disabling the modem manager?
```
$ sudo apt-get purge modemmanager
```

Kuo-Feng, 2021-06-28:
Same Issue, Is this issue fixed or not?

ROBOTIS-Will, 2021-06-29:
@Kuo-Feng 
Please give us more information about your system:
- Which OS you installed on SBC?
- Which ROS version are you using?
- Which RPi model do you have?
- Have you tried with a provided SBC image from ROBOTIS eManual?

Kuo-Feng, 2021-07-01:
> @Kuo-Feng
> Please give us more information about your system:
> 
> * Which OS you installed on SBC?
> * Which ROS version are you using?
> * Which RPi model do you have?
> * Have you tried with a provided SBC image from ROBOTIS eManual?

I installed Ubuntu 20.04 on SBC, from the image that provided on the tutorials. And the ROS version is noetic. The RPi model is the one that Turtlebot 3 burger used.

There will have that message, but I think the firmware is uploaded to OpenCR sucessfully, because I can remote control the burger through teleop_key package.

Not sure that failure will have impact or not.
But currently, it works just fine.

Thank you.

ROBOTIS-Will, 2021-07-02:
@Kuo-Feng 
If Raspberry Pi fails to access the ttyACM0 port to communicate with OpenCR, it could cause malfunction in operating TurtleBot3.
Since you have successfully operated with teleop_key node, it looks fine, but the cause of the problem looks like the modemmanager as described in the RPi forum below.
https://raspberrypi.stackexchange.com/questions/102022/bad-serial-connection-on-usb-otg-serial-on-pi-zero
Thank you.

Lopetex, 2021-12-07:
Same Issue here on TurtleBot3 Burger ROS2 (FOXY)
It says [OK] jump_to_fw, can I proceed, or I will get some problems in the future?

keerthanamanivannan, 2022-05-20:
@Lopetex I'm also getting the same error. Did you find a fix? or did you not face any other issues related to this?

Lopetex, 2022-05-23:
Hi @keerthanamanivannan, I'm sorry but I couldn't find a fix, but as far as I know this message might not be crucial to make the robot work. In my experience, I kept encountering problems working with ROS2 and I arrived to one dead point where I couldn't communicate with the robot. The communication system changes between ROS and ROS2 so I tried ROS Noetic instead and worked amazing.
Good luck!