ROBOTIS-GIT/turtlebot3 #742: No map received in Rviz using Turtlebot3 in Gazebo- Reopen issue

nagarjunvinukonda, 2021-05-13:
Hi, I am facing the same problem. can this issue be reopened?

I am using ROS melodic 18.0.4.

I am trying to gmap the turtlebot3_world.launch file using the command:
`roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping 
`
But, I am having trouble with **No map received:**

As  @routiful  told above I dont find any mistake in my terminal while I launch gmapping, T**His is the following output in my terminal :**

`
nagarjunv@nagarjunv-Inspiron-7580:~/hk_ws$ roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping 
... logging to /home/nagarjunv/.ros/log/891ec6d0-b42e-11eb-90d9-3c6aa7417fe4/roslaunch-nagarjunv-Inspiron-7580-12859.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://nagarjunv-Inspiron-7580:38227/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /turtlebot3_slam_gmapping/angularUpdate: 0.2
 * /turtlebot3_slam_gmapping/astep: 0.05
 * /turtlebot3_slam_gmapping/base_frame: base_footprint
 * /turtlebot3_slam_gmapping/delta: 0.05
 * /turtlebot3_slam_gmapping/iterations: 5
 * /turtlebot3_slam_gmapping/kernelSize: 1
 * /turtlebot3_slam_gmapping/lasamplerange: 0.005
 * /turtlebot3_slam_gmapping/lasamplestep: 0.005
 * /turtlebot3_slam_gmapping/linearUpdate: 1.0
 * /turtlebot3_slam_gmapping/llsamplerange: 0.01
 * /turtlebot3_slam_gmapping/llsamplestep: 0.01
 * /turtlebot3_slam_gmapping/lsigma: 0.075
 * /turtlebot3_slam_gmapping/lskip: 0
 * /turtlebot3_slam_gmapping/lstep: 0.05
 * /turtlebot3_slam_gmapping/map_frame: map
 * /turtlebot3_slam_gmapping/map_update_interval: 2.0
 * /turtlebot3_slam_gmapping/maxUrange: 3.0
 * /turtlebot3_slam_gmapping/minimumScore: 50
 * /turtlebot3_slam_gmapping/odom_frame: odom
 * /turtlebot3_slam_gmapping/ogain: 3.0
 * /turtlebot3_slam_gmapping/particles: 100
 * /turtlebot3_slam_gmapping/resampleThreshold: 0.5
 * /turtlebot3_slam_gmapping/sigma: 0.05
 * /turtlebot3_slam_gmapping/srr: 0.1
 * /turtlebot3_slam_gmapping/srt: 0.2
 * /turtlebot3_slam_gmapping/str: 0.1
 * /turtlebot3_slam_gmapping/stt: 0.2
 * /turtlebot3_slam_gmapping/temporalUpdate: 0.5
 * /turtlebot3_slam_gmapping/xmax: 10.0
 * /turtlebot3_slam_gmapping/xmin: -10.0
 * /turtlebot3_slam_gmapping/ymax: 10.0
 * /turtlebot3_slam_gmapping/ymin: -10.0

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    turtlebot3_slam_gmapping (gmapping/slam_gmapping)

ROS_MASTER_URI=http://localhost:11311

process[robot_state_publisher-1]: started with pid [12894]
process[turtlebot3_slam_gmapping-2]: started with pid [12895]
process[rviz-3]: started with pid [12896]

`
The following are the images from Rviz:
[image]

I have downloaded Turtlebot Simulations and turtlebot3 packages from the following link:

https://github.com/ROBOTIS-GIT/turtlebot3_simulations
https://github.com/ROBOTIS-GIT/turtlebot3

I have freshly downloaded the above packages and catkin_make. But still don't work while doing mapping.

The following is image of rqt_graph:

[image]

THe following is image of tf_tree:

[image]

Can anyone please let me know what exactly is the problem and how to solve it?

ROBOTIS-Will, 2021-05-14:
@nagarjunvinukonda 
I just downloaded the latest source code and built and ran, but seems okay to me.
Could you give us a specific steps to reproduce your issue?
[image]

nagarjunvinukonda, 2021-05-28:
Hello ROBOTIS-Will,

Thanks for reaching out. Sorry for the late response. Yes, I have checked even with my peers, it's surprising for me why its happening for my system. Its not just for me but I have seen online this problem is happening for others too. 

I have seen the ros topics, over there the laser data is not been subscribed by gmapping package even though gazebo is publishing. Hence, the map is not been received. I dont understand why it is not subscribing.

One of the solutions I found is to reinstall the ROS. Since even though I have installed the package couple of times it still doesn't give results.

ROBOTIS-Will, 2021-05-31:
I don't see a `/tf` message in your node graph.
Could you check if you have installed tf on your machine?
```
$ sudo apt install ros-melodic-tf
```

adulouis, 2024-10-10:
I am also having the same problem even with normal rays I still have the same issue I just don't get it

GyuH13, 2024-10-31:
> I am also having the same problem even with normal rays I still have the same issue I just don't get it

Hello.
If you are experiencing the same issue, please submit a new issue following the provided template.
Thanks

closed 2024-11-08 by sunghowoo

Chillhopper, 2025-03-17:
I am having this same issue does anyone have a fix?