ROBOTIS-GIT/turtlebot3 #744: Custominzing TurtleBot3 OpenCR with one more additional Dynamixal Motor

amaheshwari2509, 2021-05-24:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [Yes ] Waffle

2. Which ROS is working with TurtleBot3?

    - [ Yes] ROS 2 Dashing Diademata

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [Yes ] Raspberry Pi 4b
    - 
4. Which OS you installed on SBC?

    - [Yes ] Ubuntu preinstalled server (18.04.5)

5. Which OS you installed on Remote PC?

    - [ Yes] Ubuntu 18.04 LTS (Bionic Beaver)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: ROS2 Dashing on
    - Firmware version: OpenCR 1.4.18 , Dynamixal2Arduno : 0.4.7

7. Specify the commands or instructions to reproduce the issue.

    -   ros2 launch turtlebot3_bringup robot.launch.py

8. Copy and Paste the error messages on terminal.

[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2021-05-24-15-01-21-107308-ubuntu-6202
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_waffle.urdf
[INFO] [robot_state_publisher-1]: process started with pid [6212]
[INFO] [hlds_laser_publisher-2]: process started with pid [6213]
[INFO] [turtlebot3_ros-3]: process started with pid [6214]
[robot_state_publisher-1] Initialize urdf model from file: /home/ubuntu/turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_waffle.urdf
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_link had 7 children
[robot_state_publisher-1] Link camera_link had 2 children
[robot_state_publisher-1] Link camera_depth_frame had 1 children
[robot_state_publisher-1] Link camera_depth_optical_frame had 0 children
[robot_state_publisher-1] Link camera_rgb_frame had 1 children
[robot_state_publisher-1] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher-1] Link caster_back_left_link had 0 children
[robot_state_publisher-1] Link caster_back_right_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[robot_state_publisher-1] got segment base_footprint
[robot_state_publisher-1] got segment base_link
[robot_state_publisher-1] got segment base_scan
[robot_state_publisher-1] got segment camera_depth_frame
[robot_state_publisher-1] got segment camera_depth_optical_frame
[robot_state_publisher-1] got segment camera_link
[robot_state_publisher-1] got segment camera_rgb_frame
[robot_state_publisher-1] got segment camera_rgb_optical_frame
[robot_state_publisher-1] got segment caster_back_left_link
[robot_state_publisher-1] got segment caster_back_right_link
[robot_state_publisher-1] got segment imu_link
[robot_state_publisher-1] got segment wheel_left_link
[robot_state_publisher-1] got segment wheel_right_link
[hlds_laser_publisher-2] [INFO] [hlds_laser_publisher]: Init hlds_laser_publisher Node Main
[hlds_laser_publisher-2] [INFO] [hlds_laser_publisher]: port : /dev/ttyUSB0 frame_id : base_scan
[ERROR] [hlds_laser_publisher-2]: process has died [pid 6213, exit code 255, cmd '/opt/ros/dashing/lib/hls_lfcd_lds_driver/hlds_laser_publisher __node:=hlds_laser_publisher __params:=/tmp/launch_params_xeeaz84i'].
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Init DynamixelSDKWrapper
[robot_state_publisher-1] Adding fixed segment from base_footprint to base_link
[robot_state_publisher-1] Adding fixed segment from base_link to camera_link
[robot_state_publisher-1] Adding fixed segment from camera_link to camera_depth_frame
[robot_state_publisher-1] Adding fixed segment from camera_depth_frame to camera_depth_optical_frame
[robot_state_publisher-1] Adding fixed segment from camera_link to camera_rgb_frame
[robot_state_publisher-1] Adding fixed segment from camera_rgb_frame to camera_rgb_optical_frame
[robot_state_publisher-1] Adding fixed segment from base_link to caster_back_left_link
[robot_state_publisher-1] Adding fixed segment from base_link to caster_back_right_link
[robot_state_publisher-1] Adding fixed segment from base_link to imu_link
[robot_state_publisher-1] Adding fixed segment from base_link to base_scan
[robot_state_publisher-1] Adding moving segment from base_link to wheel_left_link
[robot_state_publisher-1] Adding moving segment from base_link to wheel_right_link
[turtlebot3_ros-3] [INFO] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[turtlebot3_ros-3] [ERROR] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-3]   what():  could not create publisher: rcl node's context is invalid, at /tmp/binarydeb/ros-dashing-rcl-0.7.9/src/rcl/node.c:476
[ERROR] [turtlebot3_ros-3]: process has died [pid 6214, exit code -6, cmd '/home/ubuntu/turtlebot3_ws/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 __params:=/home/ubuntu/turtlebot3_ws/install/turtlebot3_bringup/share/turtlebot3_bringup/param/waffle.yaml'].

9. Please describe the issue in detail.

    - I am customising the Turtlebot 3 with new additional Dynamixal motor to be added in front along with rear 2 wheels. I am using ROS 2 Turtlebot (https://github.com/ROBOTIS-GIT/turtlebot3/tree/ros2) and OpenCR Turtlebot3 ROS 2 Firmware (https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2). I have further changed (customised) the motor_driver.cpp and motor_driver.h file where Dynamixal2Arduino library is used to add the new motor id and other details such as functions. Changes are done in following file : https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/src/turtlebot3/turtlebot3_motor_driver.cpp. For the third motor (named as centre) I have added following code in the file : 
------------------
 const uint8_t DXL_MOTOR_ID_FRONT
  sync_write_param.id_count = 3;                 //**
  sync_write_param.xel[LEFT].id = left_wheel_id_;
  sync_write_param.xel[RIGHT].id = right_wheel_id_;
  sync_write_param.xel[CENTER].id = front_wheel_id_;    //**
------------
  sync_read_param.addr = 132;
  sync_read_param.length = 4;
  sync_read_param.id_count = 3;
  sync_read_param.xel[LEFT].id = left_wheel_id_;
  sync_read_param.xel[RIGHT].id = right_wheel_id_;
  sync_read_param.xel[CENTER].id = front_wheel_id_;    //**
--------------
bool Turtlebot3MotorDriver::set_torque(bool onoff)
{
  bool ret = false;

  sync_write_param.addr = 64;
  sync_write_param.length = 1;
  sync_write_param.xel[LEFT].data[0] = onoff;
  sync_write_param.xel[RIGHT].data[0] = onoff;
  sync_write_param.xel[CENTER].data[0] = onoff;          //**

  if(dxl.syncWrite(sync_write_param) == true){
    ret = true;
    torque_ = onoff;
  }

  return ret;
}
------------
bool Turtlebot3MotorDriver::write_position(int32_t center_value)
{
  bool ret = false;

  sync_write_param.addr = 116;
  sync_write_param.length = 4;
  memcpy(sync_write_param.xel[CENTER].data, &center_value, sync_write_param.length);

  if(dxl.syncWrite(sync_write_param)){
    ret = true;
  }

  return ret;
}
---------------
// front_joint_value is the function of linear velocity and angular velocity coming from cmd_vel
dxl_comm_result= write_position(front_joint_value);
-------------------    

    - While uploading the firmware OpenCR shows no compilation error and new Dynamixal motor is also initialised but while launching the ROS2 node from robot.launch.py (from Turtlebot 3 ROS2 Library) the errors is displayed as shown above in section 8. Furthermore, to eliminate this error if I write following commands : 
    - 
$ export OPENCR_PORT=/dev/ttyACM0
$ export OPENCR_MODEL=burger
$ rm -rf ./opencr_update.tar.bz2
$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2 
$ tar -xvf opencr_update.tar.bz2 
$ cd ./opencr_update
$ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr

Then the error is eliminated but third motor doesn't work only other two motors work properly. I am sending the cmd_vel values from MATLAB which is also on ROS2 Network. 

Can anyone over here please explain how I can solve the error shown above and make third motor work properly. Do I need to change the code in turtlebot3.cpp file also and is any change also needed from ROS2 node files ?

Thanks

ROBOTIS-Will, 2021-05-26:
For ROS2 TurtleBot3, OpenCR works as a slave device like DYNAMIXEL.
OpenCR creates its own control table which consists of memory address and data byte in each address as below.
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/src/turtlebot3/turtlebot3.cpp#L109
Using this control table, OpenCR and the SBC such as Raspberry Pi communicate to each other with DYNAMIXEL Protocol 2.0.

In order to add another DYNAMIXEL module to the platform, you'll need to create a specific control table address that will represent the specific item of the DYNAMIXEL such as Goal Position and Present Position, then use these values to control DYNAMIXEL from Raspberry Pi.
[image]

amaheshwari2509, 2021-05-26:
Thanks for your reply. I will work on this issue and will close this as soon as it is solved.

amaheshwari2509, 2021-06-18:
I have been able to connect the OpenCR with customised dynamixel motor (third motor) but still facing issues to perform communication between OpenCR and Raspberry PI Turtlebot 3 main node.  I can now control motor by giving input in the OpenCR (motor_driver.cpp) while uploading OpenCR turtlebot3 firmware but unable to control it from inputs from the ROS node (/cmd_vel topic Twist commands). When I launch "bringup" I am getting following error  "Unable to connect devices"  :

Figure 1
[image]

Following are the changes (presented in concise manner) I did, while working on this issue : 

**Task I : OpenCR - Dynamixel Interface** (working fine)

_1. MotorDriver Dynamixal Control Table_ (changes in OpenCR/Libraries/Turtlebot3_ros2/Turtlebot3_motor_drover .h &  .cpp ) 

Figure 2
[image]

_2.  Open CR Control Table & Callback_ (changes in OpenCR/Libraries/Turtlebot3_ros2/Turtlebot3 .cpp ) 

Figure 3
[image]

Post first and second step after uploading Turtlebot3_waffle.ino file on OpenCR, new dynamixel motor is running properly based on inputs to linear value and angular value inputs. 

**Task II : Rapberry Pi - OpenCR  Interface** (query !!! error reference from figure 1 : Device not added)

_3.  Turtlebot3 ROS2 & Dynamixel Wrapper (changes in Turtlebot3_ros2/Turtlebot3_node/Turtlebot3 .cpp & ControlTable.hpp) 

I have added same address of control item parameters in controltable.hpp for 3rd motor same as  added in OpenCR in figure 2. After colcon build, i am recieving error "" Device not added" which is related to TurtleBot3::check_device_status() and further to Dynamic SDK Wrapper bool DynamixelSDKWrapper::read_register. I would be thankful if you can suggest me what i am missing in this process which is causing that error ? Do I need to do some changes in Dynamixal SDK Wrapper.cpp code ? Is 3rd Motor device_id is not getting register causing broken communication ?  

Figure 4
[image]

Thank you for taking time to solve this error.

-----------------------------------------------------------------------------------------------------------------
> For ROS2 TurtleBot3, OpenCR works as a slave device like DYNAMIXEL.
> OpenCR creates its own control table which consists of memory address and data byte in each address as below. [...]

ROBOTIS-Will, 2021-06-22:
Hi @amaheshwari2509 
Please understand that we do not provide a thorough support on customizing the source code.
I'm not sure how your system is designed, but assuming that `cmd_vel` is used to drive 2 DYNAMIXEL modules and to steer the third module.
I'll try to write an example regarding adding a third DYNAMIXEL to TurtleBot3, but please allow me some time as I have other imminent tasks to be done by this month.
I believe you are almost getting it, and just missing a few minor things.

amaheshwari2509, 2021-06-30:
> Hi @amaheshwari2509
> Please understand that we do not provide a thorough support on customizing the source code.
> I'm not sure how your system is designed, but assuming that `cmd_vel` is used to drive [...]

Sure, I acknowledge the same. Yes 'cmd_vel' is used for all the three motors (2 propulsion + 1 steering). Will look forward to your response in future. In meantime, I will also try to resolve "opencr-raspberrypi ros2 node communication" issue by myself.  Thank again for your time to guide through this issue.

ROBOTIS-Will, 2021-06-30:
Thanks for your understanding.
I should be able to put some time for this before the mid of July.
Please feel free to update any progress via this thread.

amaheshwari2509, 2021-07-14:
Hello @ROBOTIS-Will , hope you doing good. Did you got time to look into above issue ? As from my understanding for this issue, Rpi-OpenCR are unable to communicate which can be acknowledge from "Failed connection with devices." while launching. While debugging, I tracked for the function  call which might cause this i.e. check_device_status() , is_connected_to_devices() and to read_register() but unable to solve the same. Thanks a lot again.

ROBOTIS-Will, 2021-07-19:
@amaheshwari2509 
While I was testing the Arduino code last week, I also experienced the OpenCR does not communicate well with the RPi and getting the connection failure when bringup.
It looks like the control table created in the OpenCR does not respond to the DYNAMIXEL instruction from the Raspberry Pi.
I'll investigate more on this and get back to you.
Sorry about your inconvenience.

amaheshwari2509, 2021-07-19:
@ROBOTIS-Will 

I would like to thank you for looking into this issue in detail. I acknowledge your feedback on this bug issue. I will look forward to hear from you once it is solved from your side. Meanwhile, I  will also keep digging into this issue. If you could suggest me any particular bit to look out for , please let me know  and I shall also keep digging into this issue to resolve this. 

Kind Regards

ROBOTIS-Will, 2021-07-21:
Hi @amaheshwari2509 

Thank you for your help on this.

The Control Table may not be properly created in the OpenCR as I cannot detect ID 200 when I tried scanning the OpenCR with Wizard 2.0 through dev/ttyACM0 port. Or OpenCR may not even responding to the DYNAMIXEL Protocol.

Another possible cause of the issue is that the [DynamixelSDKWrapper::read_register()](https://github.com/ROBOTIS-GIT/turtlebot3/blob/foxy-devel/turtlebot3_node/src/dynamixel_sdk_wrapper.cpp#L118) in the dynamixel_sdk_wrapper.cpp fails retrieving the RX packet from the OpenCR. I still need to investigate on this, but I'm guessing that the problem is mainly caused by the OpenCR side.

amaheshwari2509, 2021-08-28:
Hello @ROBOTIS-Will 

Has the bug related to RPi - OpenCR communication for ROS2 has been solved ? Looking forward to your reply. 

Thanks

> Hi @amaheshwari2509
>
> Thank you for your help on this.
>
> The Control Table may not be properly created in the OpenCR as I cannot detect ID 200 when I tried scanning the OpenCR with Wizard 2.0 through [...]

ROBOTIS-Will, 2021-08-30:
Hi @amaheshwari2509 
Thank you for the friendly reminder on this issue.
I'm still working on this issue found some discrepancy between the binary and the source code since 2019, but wasn't able to spend enough time to dig deeper due to an imminent hardware update related to TurtleBot3.
I'll keep you updated if I find something helpful to resolve the issue.
Thanks!

ROBOTIS-Will, 2021-09-06:
@amaheshwari2509 
A small update while we're keep working on this.
Using the `OpenCR Arduino Board Manager (1.4.12)` with `DYNAMIXEL2Arduino library (0.3.0)` helps to resolve the issue.
Below fixes should be applied however.

1. [modify the platform.txt file](https://github.com/ROBOTIS-GIT/OpenCR/blob/92b7286aa1c73a765fc023aebb35840978d3b2d9/arduino/opencr_arduino/opencr/platform.txt#L68-L73 ) as below.
```
## Compile c files
recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.c.flags} -mcpu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.c.extra_flags} {build.extra_flags} {build.cpu_flags} {build.common_flags} {compiler.libs.c.flags} {includes} "{source_file}" -o "{object_file}"
###

## Compile c++ files
recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} -mcpu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {build.cpu_flags} {build.hs_flag} {build.common_flags} {compiler.libs.c.flags} {includes} "{source_file}" -o "{object_file}"
```

2. [Add TORQUE_ENABLE definition in the turtlebot3_motor_driver.h file](https://github.com/ROBOTIS-GIT/OpenCR/blob/92b7286aa1c73a765fc023aebb35840978d3b2d9/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/include/turtlebot3/turtlebot3_motor_driver.h#L20 ) as below.
```
#include <Dynamixel2Arduino.h>

#define TORQUE_ENABLE 1
```

amaheshwari2509, 2021-09-07:
> @amaheshwari2509
> A small update while we're keep working on this.
> Using the `OpenCR Arduino Board Manager (1.4.12)` with `DYNAMIXEL2Arduino library (0.3.0)` helps to resolve the issue.
> Below fixes should be applied however.
> 
> 1. [modify the platform.txt file](https://github.com/ROBOTIS-GIT/OpenCR/blob/92b7286aa1c73a765fc023aebb35840978d3b2d9/arduino/opencr_arduino/opencr/platform.txt#L68-L73) as below.
> 
> ```
> ## Compile c files
> recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.c.flags} -mcpu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.c.extra_flags} {build.extra_flags} {build.cpu_flags} {build.common_flags} {compiler.libs.c.flags} {includes} "{source_file}" -o "{object_file}"
> ###
> 
> ## Compile c++ files
> recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} -mcpu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {build.cpu_flags} {build.hs_flag} {build.common_flags} {compiler.libs.c.flags} {includes} "{source_file}" -o "{object_file}"
> ```
> 
> 1. [Add TORQUE_ENABLE definition in the turtlebot3_motor_driver.h file](https://github.com/ROBOTIS-GIT/OpenCR/blob/92b7286aa1c73a765fc023aebb35840978d3b2d9/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/include/turtlebot3/turtlebot3_motor_driver.h#L20) as below.
> 
> ```
> #include <Dynamixel2Arduino.h>
> 
> #define TORQUE_ENABLE 1
> ```

Thank You @ROBOTIS-Will  for your assistance in solving this issue. I will perform these changes and will test on my system. I would like to thank you for solving this issue and sharing your feedback during this process.

Best Regards

closed 2021-09-07 by amaheshwari2509