ROBOTIS-GIT/turtlebot3 #745: Turtlebot3 bringup suddenly doesn't work

sudo-Boris, 2021-05-28:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [X] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?
     I did the manual SBC setup in the emanual.

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [X] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [X] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

roslaunch turtlebot3_bringup turtlebot3_robot.launch

8. Copy and Paste the error messages on terminal.

... logging to /home/ubuntu/.ros/log/86af288a-bf01-11eb-b39e-48b02d2b81aa/roslaunch-ubuntu-2658.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt

Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://my_TURTLEBOT_IP:35829/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://my_HOST_IP:11311

process[turtlebot3_core-1]: started with pid [2667]
process[turtlebot3_lds-2]: started with pid [2668]
process[turtlebot3_diagnostics-3]: started with pid [2669]
[INFO] [1622135149.303476, 0.000000]: ROS Serial Python Node
[INFO] [1622135149.398551, 0.000000]: Connecting to /dev/ttyACM0 at 115200 baud

9. Please describe the issue in detail.

    I have worked with the turtlebot without issues, but at one seemingly random point I could not run the bringup of the 
    turtlebot.

    It stops when trying to connect to /dev/ttyACM0.

    Other post mentioned it was a battery issue. I charged my turtlebot3 up to full. Then the next bringup worked. But after that 
    one bringup, it had the same error again, even after again charging those few percentages.

     I have tried doing an apt-get update and upgrade -> still the same issue. What could it be? All the sudden stopped 
     working..

usfmel, 2021-05-30:
I also had the very same problem; it seemed to disappear (temporarily) by executing

 rosrun turtlebot3_bringup create_udev_rules

    ...which is step 15 in the manual setup described in

https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup

   ...and has something to do with USB setup. You have to "expand" the text at the bottom of the page to see it.  However, the problem reappeared shortly afterwards, after having used the robot for a few hours, and this time executing the above command did nothing to solve the problem. When I installed the software I followed the manual procedure, which was the only available by then; now you can just download a preconfigured image. I will try this when I have time....

sudo-Boris, 2021-05-30:
I actually did the manual SBC setup and also tried to rerun the command 
`rosrun turtlebot3_bringup create_udev_rules`.
One time it worked, the next time it didn't. I now tried it again and it did work again. Let's see how long this will hold on this time.
But thanks for reminding me to try it again anyway!

ROBOTIS-Will, 2021-05-31:
@sudo-Boris 
Have you also tried rebooting the OpenCR board with the `Reset` button before launching the turtlebot3_robot node?

sudo-Boris, 2021-05-31:
First I tried to reupload the firmare with `./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr`.
Then I pressed down the reset for a few seconds.
now the error is that the port can not be found.
`[ERROR] [1622441529.074521, 0.000000]: Error opening serial: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
`

ROBOTIS-Will, 2021-05-31:
@sudo-Boris 
In case of failing the firmware upload, please refer to the `recovery mode` below instead of resetting the board.
https://emanual.robotis.com/docs/en/parts/controller/opencr10/#recovery-mode

sudo-Boris, 2021-06-02:
It wasn't necessary, it disappeared with the second try.
But what I found as a consistent fix is **restarting your roscore** after the bringup fails.
I think it could be a synchronisation issue, but at least it seems to work.

closed 2024-11-08 by sunghowoo