ROBOTIS-GIT/turtlebot3 #747: protocol mismatch

usaotearoa, 2021-06-05:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?
    - [X] Burger SELF BUILT with different dynamixel

2. Which ROS is working with TurtleBot3?
    - [X] ROS 1 Noetic Ninjemys

3. Which SBC(Single Board Computer) is working on TurtleBot3?
- [X] Raspberry Pi 4

4. Which OS you installed on SBC?
    - [X] Ubuntu (20.04)

5. Which OS you installed on Remote PC?
- [X] Ubuntu 20.04 LTS (Focal Fossa)

6. Specify the software and firmware version(Can be found from Bringup messages)
    - Hardware version: [0.0.0]
    - Software version: [1.2.6]
    - Firmware version: [0.0.0]   // I'm using customized arduino sketch see below

7. Specify the commands or instructions to reproduce the issue.
This is case specific, only applies to my build

8. Copy and Paste the error messages on terminal.
# ===========================================
# roslaunch turtlebot3_bringup turtlebot3_robot.launch
# ===========================================
```
... logging to /home/ubuntu/.ros/log/f6a67018-c563-11eb-9e85-738df4336004/roslaunch-ubuntu-8512.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.111:37071/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)

ROS_MASTER_URI=http://192.168.1.101:11311

process[turtlebot3_core-1]: started with pid [8521]
process[turtlebot3_diagnostics-2]: started with pid [8522]
[INFO] [1622841780.228738]: ROS Serial Python Node
[INFO] [1622841780.298141]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1622841782.423497]: Requesting topics...
[INFO] [1622841783.139064]: Note: publish buffer size is 1024 bytes
[INFO] [1622841783.213968]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1622841783.246273]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1622841783.327978]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1622841783.356089]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1622841783.402855]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1622841783.427447]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1622841783.444252]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1622841783.471642]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1622841783.852195]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1622841783.884633]: Note: subscribe buffer size is 1024 bytes
[INFO] [1622841783.895671]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1622841783.928405]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1622841783.961591]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1622841783.990738]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1622841785.774308]: Setup TF on Odometry [odom]
[INFO] [1622841785.795960]: Setup TF on IMU [imu_link]
[INFO] [1622841785.820182]: Setup TF on MagneticField [mag_link]
[INFO] [1622841785.838468]: Setup TF on JointState [base_link]
[ERROR] [1622841785.858865]: Tried to publish before configured, topic id 131
[INFO] [1622841785.878074]: Requesting topics...
[ERROR] [1622841785.922374]: Tried to publish before configured, topic id 131
[INFO] [1622841785.939717]: Requesting topics...
[ERROR] [1622841785.990597]: Tried to publish before configured, topic id 131
[INFO] [1622841786.008171]: Requesting topics...
[ERROR] [1622841786.060857]: Tried to publish before configured, topic id 131
[INFO] [1622841786.076387]: Requesting topics...
[ERROR] [1622841786.135769]: Tried to publish before configured, topic id 131
[INFO] [1622841786.164970]: Requesting topics...
[ERROR] [1622841786.244816]: Tried to publish before configured, topic id 131
[INFO] [1622841786.255866]: Requesting topics...
[ERROR] [1622841786.352616]: Tried to publish before configured, topic id 131
[INFO] [1622841786.377167]: Requesting topics...
[ERROR] [1622841786.420876]: Tried to publish before configured, topic id 131
[INFO] [1622841786.434314]: Requesting topics...
[ERROR] [1622841786.498712]: Tried to publish before configured, topic id 131
[INFO] [1622841786.521882]: Requesting topics...
[INFO] [1622841786.626478]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1622841786.644568]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
```

# ===========================================
# rostopic echo /diagnostics 
# ===========================================
```
header: 
  seq: 21
  stamp: 
    secs: 1622841800
    nsecs: 831201144
  frame_id: ''
status: 
  - 
    level: 0
    name: "IMU Sensor"
    message: "Good Condition"
    hardware_id: "MPU9250"
    values: []
  - 
    level: 0
    name: "Actuator"
    message: "Torque ON"
    hardware_id: "DYNAMIXEL X"
    values: []
  - 
    level: 0
    name: ''
    message: ''
    hardware_id: ''
    values: []
  - 
    level: 0
    name: "Power System"
    message: "Good Condition"
    hardware_id: "Battery"
    values: []
  - 
    level: 0
    name: "Analog Button"
    message: "Pushed Nothing"
    hardware_id: "OpenCR Button"
    values: []
---
header: 
  seq: 1
  stamp: 
    secs: 1622841801
    nsecs:  73665142
  frame_id: ''
status: 
  - 
    level: 2
    name: "rosserial_python"
    message: "Mismatched protocol version in packet: lost sync or rosserial_python is from different\
  \ ros release than the rosserial client"
    hardware_id: ''
    values: 
      - 
        key: "last sync"
        value: "Fri Jun  4 21:23:20 2021"
      - 
        key: "last sync lost"
        value: "Thu Jan  1 00:00:00 1970"
---
```

9. Please describe the issue in detail.
Hi all,

I have built my own version of the turtlebot with the same hardware, just 3D printed materials. I’m getting a lot of errors when running “roslaunch turtlebot3_bringup turtlebot3_robot_launch”

STOP: Before you suggest to follow the manual and reflash the firmware, please continue reading !!!!

I have different hardware and specs, for example, my tire size is different and my turn_radius is different, as well as some other stuff. Therefore I had to modify the definitions found in
OpenCR/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_burger.h
```
#ifndef TURTLEBOT3_BURGER_H_
#define TURTLEBOT3_BURGER_H_

#define NAME                             "Burger"

#define WHEEL_RADIUS                     0.0356          // meter                                       // 35.6mm
#define WHEEL_SEPARATION                 0.172967        // meter (BURGER : 0.160, WAFFLE : 0.287)      // 172.967mm
#define TURNING_RADIUS                   0.086483        // meter (BURGER : 0.080, WAFFLE : 0.1435)     // 86.4835
#define ROBOT_RADIUS                     0.245           // meter (BURGER : 0.105, WAFFLE : 0.220)      //
#define ENCODER_MIN                      -2147483648     // raw                                         //
#define ENCODER_MAX                      2147483648      // raw                                         //
#define MAX_LINEAR_VELOCITY              (WHEEL_RADIUS * 2 * 3.14159265359 * 46 / 60) // m/s  (BURGER: 61[rpm], WAFFLE: 77[rpm], XM430-W350-T: 46[rpm])       //
#define MAX_ANGULAR_VELOCITY             (MAX_LINEAR_VELOCITY / TURNING_RADIUS)       // rad/s
#define MIN_LINEAR_VELOCITY              -MAX_LINEAR_VELOCITY  
#define MIN_ANGULAR_VELOCITY             -MAX_ANGULAR_VELOCITY 
#endif  //TURTLEBOT3_BURGER_H_
```

Due to these customisations, I have to load the compiled sketch via arduino interface onto the openCR board, then plug it into my raspberry pi 4b. 
I believe the root cause seems to be a version difference between the fw running on my openCR and the turtlebot3 package version used on the raspberry pi.
I get the below console output when attempting to run everything.

SEE CONSOLE OUTPUT ABOVE

ROS master communication is set-up properly and works. Well even though I get all these error messages, it still works and is publishing data properly and I can move the bot with teleop_twist. However the error output in the console window, where I did the roslaunch bringup… is concerning and spamming the entire screen.
Monitoring topics with rostopic list -v and then displaying with rostopic echo works just fine as well. I did notice the following in the diagnostic output:
"Mismatched protocol version in packet: lost sync or rosserial_python is from different\
  \ ros release than the rosserial client"

SEE ABOVE FOR DETAILS ON DIAGNOSTICS

Can you please help me get rid of these ERRORS?

usaotearoa, 2021-06-05:
Hi all,

I just wanted to update on the issue and how it was solved in case someone else has the same problem.
Further investigation into the issue has show that both the node_handle.h on the arduino side (OpenCR) and the SerialClient in rosserial_python are using the correct protocol version, as they are set to the same.
See extract below

`C:\Users\User01\Documents\GitHub\OpenCR\arduino\opencr_arduino\opencr\libraries\turtlebot3_ros_lib\ros\node_handle.h`

```
/*
 * The second sync byte is a protocol version. It's value is 0xff for the first
 * version of the rosserial protocol (used up to hydro), 0xfe for the second version
 * (introduced in hydro), 0xfd for the next, and so on. Its purpose is to enable
 * detection of mismatched protocol versions (e.g. hydro rosserial_python with groovy
 * rosserial_arduino. It must be changed in both this file and in
 * rosserial_python/src/rosserial_python/SerialClient.py
 */
const uint8_t MODE_PROTOCOL_VER   = 1;
const uint8_t PROTOCOL_VER1       = 0xff; // through groovy
const uint8_t PROTOCOL_VER2       = 0xfe; // in hydro
const uint8_t PROTOCOL_VER        = PROTOCOL_VER2;
const uint8_t MODE_SIZE_L         = 2;
const uint8_t MODE_SIZE_H         = 3;
const uint8_t MODE_SIZE_CHECKSUM  = 4;    // checksum for msg size received from size L and H
const uint8_t MODE_TOPIC_L        = 5;    // waiting for topic id
const uint8_t MODE_TOPIC_H        = 6;
const uint8_t MODE_MESSAGE        = 7;
const uint8_t MODE_MSG_CHECKSUM   = 8;    // checksum for msg and topic id

```

```
class SerialClient(object):
    """
        ServiceServer responds to requests from the serial device.
    """
    header = b'\xff'

    # hydro introduces protocol ver2 which must match node_handle.h
    # The protocol version is sent as the 2nd sync byte emitted by each end
    protocol_ver1 = b'\xff'
    protocol_ver2 = b'\xfe'
    protocol_ver = protocol_ver2
```
	

So the mismatch had to come from somewhere. while browsing though the code. I noticed the following line was commented out and after
uncommenting it, compiling and downloading to the openCR board, everything works as expected :-)	

`#define NOETIC_SUPPORT          //uncomment this if writing code for ROS1 Noetic`

CLOSED THIS ISSUE

closed 2021-06-05 by usaotearoa

ROBOTIS-Will, 2021-06-07:
Hi @usaotearoa 
Just in case you might have missed our eManual, please see below page.
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup
Make sure to select `Noetic` on top of the page to switch the content for ROS1 Noetic.
Thank you!