ROBOTIS-GIT/turtlebot3 #749: LDS publishes no data on /scan topic

ArkenSmaug, 2021-06-12:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [x ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - I was working with my turtlebot and everything was running fine. Was able to visualize the surroundings using rviz and read the data from the lds vis the scan topic. However, returning after some time, even though the Lidar rotates just fine, there appears to be no data on the scan topic. Additionally, there seems to be no LDS/Lidar on the diagnostics topic.

ROBOTIS-Will, 2021-06-14:
@ArkenSmaug 
If the time difference between your PC and SBC gets too big, messages could be ignored.
You might need to run timesync for both PC and SBC.
https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc

ArkenSmaug, 2021-06-15:
@ROBOTIS-Will I tried that. But it didn't help. 
Incidentally, when I re-ran the next day, everything seemed to work just fine. Since then I have faced this issue once again.
Not sure why but maybe I am missing out on something?

ROBOTIS-Will, 2021-06-16:
There could be several reasons of LDS issues.
It could be a network issue or timesync.
Sometimes the hardware such as USB2LDS or USB cable or LDS itself could be an issue. In this case, you can connect your LDS to your PC and test the sensor data.
https://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_lds_01/#user-guide-for-ros

ArkenSmaug, 2021-06-17:
Ok, but what bugs me here is that the error is not persistent. It appears to happen at seemingly unrelated times, and fixes itself on its own accord.

ArkenSmaug, 2021-06-30:
I tried connecting the lds to my pc on vm linux machine. Tried the .make and then ./lds_driver command and got the following output.

`make
g++ -o lds_driver lds_driver.o -I/usr/include -L/usr/lib/x86_64-linux-gnu -lboost_system -lpthread

./lds_driver 
An exception was thrown: open: Permission denied`.

Havent used this lds/usb converter so im not sure if there is an error or if the vm is simply rejecting some connection. 

Although when I try the lsusb on the vm terminal I am able to see:

`Bus 002 Device 006: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light`

And for dmesg | grep tty command

`cp210x converter now attached to ttyUSB0`

ROBOTIS-Will, 2021-07-01:
@ArkenSmaug 
It could be you don't have a permission to access the USB port from your VM.
I'm not sure how you connected the LDS-01 to your PC without the USB2LDS, but please understand that I'm not able to provide a thorough support if the TurtleBot3 is running under customized hardware and software environment as there are way too many possible combinations.
Thanks!

ArkenSmaug, 2021-07-01:
@ROBOTIS-Will I did use the USB2LDS, to connect the lds to my pc. It went LDS->USB2LDS->PC(VM).
I'd like to re-emphasize the point of it working at times and not working at other times, seemingly randomly. Because after the pc errors I reconnected it to the on-board Rpi and was getting the data on the topic. 

I'm sure that there must be a reason for this sporadic behaviour.

ROBOTIS-Will, 2021-07-02:
@ArkenSmaug 
I'd recommend to try it on a native Linux machine as it is very difficult to find the cause of problem when running on VM.

ArkenSmaug, 2021-07-13:
Ok, I installed the requisites on the onboard RPi itself. 

Running the lsusb command I can see 
[image]

Clearly Device 6 is attached to the rpi and I take it that it is the USB2LDS. 

However when I make the lds driver in the hls_lfcd_lds_driver/applications/lds_driver/  folder, and running ./lds_driver gives no output.

Didn't respond till now, because again the lds was working for a few days in between there.

closed 2021-07-13 by ArkenSmaug