ROBOTIS-GIT/turtlebot3 #750: Overwriting odometry

paolorugg, 2021-06-13:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [X] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [X] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - For my project, I was trying to exploit a landmark of known location to impose a precise pose to turtlebot3, thus nullifying the odometry error that naturally accumulates during motion. I have built a ros2 node that publishes an odometry message on /odom topic, and it works, but the problem is that when it stops sending messages to /odom, the pose on /odom goes back to the one estimated ignoring my input. In my idea instead it should go on estimating its pose but starting from the pose i provided. 
    I have also searched for a ros2 service to be used, but couldn't find any.  Maybe there is something I'm missing on how turtlebot3 provides its position. If you could help me on this, it would be really appreciated.

ROBOTIS-Will, 2021-06-14:
Hi @paolorugg 
The odometry of TurtleBot3 in ROS2 is calculated in the `turtlebot3_ros` node as below at the startup of `robot.launch.py`.
https://github.com/ROBOTIS-GIT/turtlebot3/blob/foxy-devel/turtlebot3_node/src/odometry.cpp#L217

Since `robot_pose_[]` is a private variable of the `Odometry` class, you'll need to create a new function in order to update the value.

paolorugg, 2021-06-14:
Thank you for the fast reply. I have given a look at the code, and now my question is: will it be enough for my function to over write robot_pose_[] and reset to 0 values diff_joint_position[], or are there more variables to be reset? Because for the moment I would liker to change only the position of the robot, and keep the estimated orientation. Thanks again.

ROBOTIS-Will, 2021-06-14:
The orientiation isn't affected by the position of the robot as below.
https://github.com/ROBOTIS-GIT/turtlebot3/blob/foxy-devel/turtlebot3_node/src/odometry.cpp#L155
https://github.com/ROBOTIS-GIT/turtlebot3/blob/foxy-devel/turtlebot3_node/src/odometry.cpp#L261

paolorugg, 2021-06-16:
I worked a bit and changed odometry.cpp and odomety.hpp. What I added is a subscriber to my topic "/pose_relocalization" where a geometry Point message with the exact pose I would like to over write to robot_pose_ is published , only when a tag is present in the camera frame, so not continuously.  And then I over write robot_pose_ in the callback function of this new subscriber. Then I made the necessary adjustments in the headers for dependencies. Here is a link to the modified files: [https://github.com/paolorugg/my_turtlebot3_node](url). Changes are in lines 108-111 and 283-288 in odometry.cpp and lines 49, 64 and 73 in odometry.hpp.
The result is that when a tag is recognized the pose is correctly adjusted, and this time doesn't go back after stopping imposing it. However a new problem occurred. Now turtlebot3_teleop is not able to write on odom, even though it's moving the robot in the real world. The pose remains the one I imposed. Originally I thought it was due to my changes to the files, so I brought back the default files and redid a colcon build. It compiles successfully but the robot, after the initial calibration, sets a pose near to 0 and remains there forever. Turtlebot3_teleop doesn't change odom topic, that is being published continuously, just with the same exact pose. No errors nor warnings are produced in the whole process.
The two problems may be related, but I can't understand why this is happening, especially because, first, I just added a subscriber that should activate only when a message is called to the topic, so it should not interfere with the normal functioning of the robot, and second, even after going back to the original files teleop is still not updating /odom.
I hope you can help me on these issues, even though especially the second sounds a bit impossible. Thank you very much in advance

paolorugg, 2021-06-29:
Thank you very much for the help, in the end I even integrated overwriting the orientation. Much appreciated

closed 2021-06-29 by paolorugg

adbidwai, 2022-03-14:
What about overwriting odometry of Turtlebot3 if I am using ROS1?

sudar1220, 2022-06-25:
Hi @paolorugg,
Can you share your solution? I was looking for the same thing with resetting the odometery with a known pose and orientation.
I am also using ROS2 Foxy on TB3 WafflePi.
Appreciate any help in this regards,
Thanks in advance

sudar1220, 2022-06-25:
@adbidwai 
> What about overwriting odometry of Turtlebot3 if I am using ROS1?

maybe this is useful for you: https://github.com/MikeHallettUK/RosRobotics/blob/master/resetodom.py
I think this is for ROS1, but I am not sure exactly how it works, I did find this when I was looking how to reset odom in ROS2

paolorugg, 2022-06-25:
> Hi @paolorugg, Can you share your solution? I was looking for the same thing with resetting the odometery with a known pose and orientation. I am also using ROS2 Foxy on TB3 WafflePi. Appreciate any help in this regards, Thanks in advance

Yeah sure. For an immediate solution you can download my_turtlebot3_node package from my github, that is a modified version of the turtlebot3 node. Then you have to use it instead of the official one.
 I modified it to overwrite odometry by publishing a Point message (technically speaking it's a worng messaage type, but I was in the beginning) on a topic called '/pose_relocalization'. The message should be [x in meters; y in meters; angle or rotation in radians]. Once a message is sent, my modified turtlebot node reads it and over write the odometry.
Otherwise you can take inspiration and find your way of doing it, publishing a coherent message type for example.
 
 I hope this helps you!

spolstra, 2023-10-10:
Hi,

I'm new to ros and ran into the same problem setting up some turtlebots for a lab. The latest firmware update fixed the 'teleporting' problem but the turtlebots with the raspberry pi4 boards would have very large initial odometer values as already mentioned above. I noticed that some members in the `Odometry` class where not initialized so I changed the following lines in the turtlebot3_node odometry.hpp file from:
```
  std::array<double, 2> diff_joint_positions_;
  double imu_angle_;

  std::array<double, 3> robot_pose_;
  std::array<double, 3> robot_vel_;
```
to:
```
  std::array<double, 2> diff_joint_positions_ = {0.0, 0.0};
  double imu_angle_ = 0.0;

  std::array<double, 3> robot_pose_ = {0.0, 0.0, 0.0};
  std::array<double, 3> robot_vel_ = {0.0, 0.0, 0.0};
```
And then recompiling the turtlebot3_node with:
```
cd ~/turtlebot3_ws && colcon build --packages-select turtlebot3_node --symlink-install
```

Now the turtlebot always starts with values that are very close to zero, which is good enough for our lab. My assumption is that the robot_pose is used as the starting value which might explain why they are not exactly zero.

Hope this helps