ROBOTIS-GIT/turtlebot3 #754: Unable to locate turtlebot3_navigation package to chgn the paramters for the navigation

Lwk7426, 2021-06-17:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - Hi Im still new to the turtlebot3 and ROS and general. I followed the Turtlebot3 E-manual guide: https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/ to install the necessary items. However, i realized that i am unable to locate the file where turtlebot3_navigation is stored in order for me to edit the parameters. May i know do i have to download this separately? Thank you

ROBOTIS-Will, 2021-06-18:
You could find your package with the `rospack find` cli on your terminal as below.
```
$ rospack find turtlebot3_navitgation
```
If you don't see any result from the above command, make sure to source the ROS setup script.
```
$ source ~/opt/ros/kinetic/setup.bash
```
If ROS has not been installed correctly due to the recent GPG key expiration incident, please update your key as described in the below thread.
https://discourse.ros.org/t/ros-gpg-key-expiration-incident/20669
```
$ curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -
```
Thank you.

Lwk7426, 2021-06-18:
I was able to locate the file using rospack find. Thank you very much!

closed 2021-06-18 by Lwk7426