ROBOTIS-GIT/turtlebot3 #756: RLException: ERROR: unable to contact ROS master at [http://192.168.2.19:11311]

ishaanmht, 2021-06-22:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - On SBC: RLException: ERROR: unable to contact ROS master at [http://192.168.2.19:11311]

9. Please describe the issue in detail.

    -  REMOTE PC BASHRC: 
    
    export ROS_MASTER_URI=http://192.168.2.19:11311/  
    export ROS_IP=192.168.2.19

 -  SBC BASHRC: 
    
    export ROS_MASTER_URI=http://192.168.2.19:11311/  
    export ROS_IP=192.168.2.22

I have tried time syncing of machines and everything,
Strangely I am able to connect to the ros master if I set the master on rpi  and slave on my remote PC

ashekim, 2021-06-22:
Hi @ishaanmht . thank you for your question

Did you source ~/.bashrc after editing the .bashrc file like the link below?
[ROS1 Network Configuration](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#ros1-network-configuration)

ishaanmht, 2021-06-22:
Yes I am souring my bash files.

ROBOTIS-Will, 2021-06-23:
@ishaanmht 
You should use `ROS_HOSTNAME` param instead of `ROS_IP`

ishaanmht, 2021-06-23:
@ROBOTIS-Will 

I have tried that as well. It doesn't work either. 
I have noticed something strange, If I set rpi as the ros master, the remote PC is able to connect to this master. But even in this case if I publish something from my remote PC it does not reflect on the topics on rpi.

ashekim, 2021-06-24:
Hi @ishaanmht . 
Please check the ping between machines by referring to the link below.
[Basic check 2: ping between machines](https://wiki.ros.org/ROS/NetworkSetup)

medsimo, 2022-12-07:
why are u using different ROS_IP ? it should be the same Rpi IP Adress in bashrc on your laptop and RPi ... or Am i wrong ?

B-B-KING, 2023-07-14:
have you sove this? i have same problem.

GyuH13, 2024-10-31:
For those experiencing the same error, this issue is too old to address effectively. 
Please follow the updated manual, and if any further issues arise, feel free to submit a new issue.

closed 2024-11-08 by sunghowoo