ROBOTIS-GIT/turtlebot3 #757: how to add sonar to the turtlebot3

w5688414, 2021-06-23:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame

3. Which SBC(Single Board Computer) is working on TurtleBot3?

4. Which OS you installed on SBC?

    - [ ] Ubuntu MATE (16.04/18.04/20.04)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    I want to use dqn with the sonar, but I don't know how to add sonar into the turt3bolt3, Is there any tutorials or examples for me to achive my purpose?

9. Please describe the issue in detail.

    - HERE

ashekim, 2021-06-23:
Hi. @w5688414  Thanks for the question.

Please refer to the link below for related information.
[13.3.3 Ultrasonic](https://emanual.robotis.com/docs/en/platform/turtlebot3/additional_sensors/#ultrasonic)

Thank you.

ROBOTIS-Will, 2021-06-24:
Please also refer to below issue thread regarding the sonar add on to TurtleBot3
https://github.com/ROBOTIS-GIT/OpenCR/issues/233

w5688414, 2021-06-25:
I want to add sonar to TurtleBot3 in gazebo. like this, https://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#machine-learning
I want to train DQN in gazebo, not the OpenCR of TurtleBot3.

ROBOTIS-Will, 2021-06-25:
@w5688414 
YOu can refer to the link below.
https://answers.ros.org/question/362576/adding-sonar-in-gazebo-and-ros/

closed 2024-11-08 by sunghowoo