ROBOTIS-GIT/turtlebot3 #759: Adding two extra Dynamixel to the TurtleBot3 Burger

TobiasKauffer, 2021-07-09:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

6. Specify the software and firmware version(Can be found from Bringup messages)

7. Specify the commands or instructions to reproduce the issue.

8. Copy and Paste the error messages on terminal.

9. Please describe the issue in detail.

I am using a TurtleBot3 Burger with the newest ROS1 Noetic image installed. Now I want to add two Dynamixel, XL-320 and AX-12A.
The problem is that I don't know how to implement these two Dynamixels in the TurtleBot3 enviroment so they can be controlled  with rospy.

ashekim, 2021-07-12:
Hi @TobiasKauffer. Thank you for your questions.

Variable settings according to the addition of Dynamixel can be modified from the link below.
[turtlebot3 motor driver.h](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/include/turtlebot3/turtlebot3_motor_driver.h)
[turtlebot3 motor driver.cpp](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/src/turtlebot3/turtlebot3_motor_driver.cpp)

Thank you.

ROBOTIS-Will, 2021-07-12:
@TobiasKauffer 
First of all, please be aware of the input voltage as XL430 / AX-12A(11.1V) and XL-320(7.4V) have different operating voltage.
Plugging in the XL-320 to OpenCR where XL430 or AX-12A is connected at may cause hardware damage.

In regards with appending and controlling additional DYNAMIXEL module will require modification of OpenCR firmware as @ashekim mentioned above.

Thanks.

ashekim, 2021-07-14:
Hi @TobiasKauffer.
Please refer to the Dynamixel Quick Start video below.
Also, enter the `ls /dev/tty*` command to check the device name again. (Please check the 1:51 sec section of the DYNAMIXEL Quick Start Guide for ROS 1.)

[DYNAMIXEL Quick Start Guide for OpenCR 1.0](https://youtu.be/0_M0Da9SHDw)
[DYNAMIXEL Quick Start Guide for ROS 1](https://youtu.be/SpdxjsCO9sE?t=111)

Thank you.

ROBOTIS-Will, 2021-07-14:
@TobiasKauffer 
The default firmware of TurtleBot3 won't work as a communication interface between Raspberry Pi and DYNAMIXEL.
In order to run the dynamixel_workbench package to find connected DYNAMIXEL to OpenCR, you should upload the `usb_to_dxl` firmware as described in the below link.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#connect-power12v-and-ttl-or-rs485

closed 2021-07-16 by TobiasKauffer