ROBOTIS-GIT/turtlebot3 #761: turtlebot install error: 3696.517681 cdc_acm 1-1.1.2:1.0 failed to set dtr/rts at step 3.3.

laila-fangyan, 2021-07-11:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)
the one which I followed the manual to install

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.
  In 3.3.  5th step Upload firmware to the OpenCR
    - $ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr

8. Copy and Paste the error messages on terminal.

    - [3696.517681] cdc_acm 1-1.1.2:1.0 failed to set dtr/rts
    as shown in pic below
[image]

9. Please describe the issue in detail.
    - failed to set dtr/rts, couldn't enter turtlebot@turtlebot, still stays in ubuntu directory
    - I've tried solution in #738:   $ sudo apt-get purge modemmanager, also tried uploading with the recovery mode
    - neither worked.
the robot worked when I tested by Press and hold PUSH SW 1 /2
But when I try to connect turtlebot from my PC, failed.
not sure if this has something to do with the error of failed to set dtr/rts

I use ubuntu through VirtualBox in my PC, the network connection works well. 
IP of turtlebot is 172.20.10.xx
PC IP (Virtualbox) is 172.20.10.yy

I also tried to connect turtlebot3 from my pc, but after 
$ ssh ubuntu@ 172.20.10.xx
and input password: turtlebot

it returns error: 
permission denied (publickey, password)
or
connection closed by 172.20.10.xx port 22
there is eduroam wifi in my room, but I didn't use that.
I used the private wifi through my phone hotspot.
What reason could it be?

ROBOTIS-Will, 2021-07-12:
Hi @laila-fangyan 
Have you also tried setting up the USB port as below?
```
$ rosrun turtlebot3_bringup create_udev_rules
```
The OpenCR firmware download is successfully completed, but connecting to TurtleBot3 from the PC isn't related to this issue.
Virtual computing environment is not recommended as there could be various configurations that could interrupt ROS connection.

laila-fangyan, 2021-07-12:
Hi @ROBOTIS-Will 

Thanks for the instruction.
Do you mean the OpenCR setting up is completed?
I can just ignore "failed to set dtr/rts"?
Does it matter that it stays in ubuntu@ubuntu directory, not enters turtlebot@turtlebot after $ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr"?

I just tried $ rosrun turtlebot3_bringup create_udev_rules  in my PC, still returns
"permission denied".

Thanks
Fang

ROBOTIS-Will, 2021-07-13:
@laila-fangyan 
Seems like you don't have permission to run the script.
I've recovered RPi3B+ using the image file in the eManual and ran the command which resulted as below.
[image]

keerthanamanivannan, 2022-05-20:
Hi @laila-fangyan have you solved this `cdc_acm 1-1.2:1.0: failed to set dtr/rts` error?

laila-fangyan, 2022-05-20:
> Hi @laila-fangyan is this error solved `cdc_acm 1-1.2:1.0: failed to set dtr/rts`?

Hi, I gave up to install through VM.
I started over the installation under single Ubunbu OS.
And no such issue any more.

keerthanamanivannan, 2022-05-20:
@laila-fangyan I see, thank you for the reply.

sunghowoo, 2024-11-08:
This issue appears to be resolved, so it will be closed.

closed 2024-11-08 by sunghowoo