ROBOTIS-GIT/turtlebot3 #762: Navigation tutorial: possible costmap and amcl problem

azhou202, 2021-07-19:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ X] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ X] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [X ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.2.2]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    following the Robotis tutorials (Jetson is remote PC)
    - [Jetson] roscore
    - [Pi] ssh and then roslaunch turtlebot3_bringup turtlebot3_robot.launch
    - [Jetson] map was made and saved via roslaunch turtlebot3_slam turtlebot3_slam.launch and then rosrun map_server map_saver -f ~/maps/new_map
    - [Jetson] kill mapping tools and then launch navigation task via roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=/home/maps/new_map.yaml
    - [Jetson] I localize the robot using 2D Pose Estimate and teleop

8. Copy and Paste the error messages on terminal.

    NONE

9. Please describe the issue in detail.

    When I launch the navigation node I experience issues shown in rviz. When I move the robot with teleop, it takes the laser scans of the global map with it as it moves. Basically, when I move the robot with teleop, the navigation software takes the laser scans of the local map and updates the global map. Additionally, objects represented in the global map do not match up with their counterparts in the local map. For instance, rviz shows that the robot should be hitting a box (black rectangle) on the global map, but in real life and on the local map, the robot is still a distance away from the box. These two issues can be seen in the images below.

[image]

The picture above shows the robot after initial localization with 2D pose estimate and slight movements via teleop. No problems appear so far. 

[image]

The picture above shows the robot after moving from its initial position. The problem of the global map updating can be seen via the pink areas and how the laser scan representing the top wall has moved along with the robot. Additionally, the second problem can be seen in the white circles on the right. The black box is the position of the box in the global map, while the circled multi-colored area is it's position in the local map (and its position in real life). 

None of these problems occurred while doing the SLAM mapping or when using simulation. There are no error messages in terminal. Also, the issue and the images were produced by using the default params and values in the turtlebot3 repos.

To debug I have tried several things:
1. Press the reset button on the OpenCR (this suggestion was found in the thread of Issue #266 
2. I removed map_type, observation_sources, and scan from costmap_common_params_burger.yaml and placed them only in local_costmap_params.yaml because I reasoned that the global map should not be using laser data to update itself
3. Changed all mentions of base_footprint in params to base_link
4. A custom launch file with different amcl parameters

Thanks in advance for any suggestions!

ROBOTIS-Will, 2021-07-21:
@azhou202 
It looks like your map is detached from the robot.
If you have modified the code, check if each TF is connected as below.
[image]

azhou202, 2021-07-21:
@ROBOTIS-Will I checked the frames and everything seems connected. 
[image]

ROBOTIS-Will, 2021-07-22:
@azhou202 
hmm.. have you tried reinstalling the TurtleBot3 ROS packages?

azhou202, 2021-07-22:
@ROBOTIS-Will I have cloned the repo from github and placed it in my catkin_ws/src in addition to the downloads done in the Turtlebot online manual. Should I remove the packages downloaded through the steps on the tutorial and redownload?

ROBOTIS-Will, 2021-07-22:
If you have cloned the package with `git clone` and installed with `apt install`, please uninstall the binary package with `$ sudo apt uninstall ros-melodic-<name of the package>`.
As I tested yesterday with my TurtleBot3 on Melodic, I wasn't able to reproduce the issue.

azhou202, 2021-07-22:
Should I uninstall all ros packages or just turtlebot3-msgs and turtlebot3?

azhou202, 2021-07-22:
I uninstalled ros-melodic-turtlebot3-msgs and ros-melodic-turtlebot3 with `sudo apt-get autoremove ros-melodic-{package name} `and reinstalled using the tutorial's instructions. I tested it out, and it is still doing the same thing as before.

ROBOTIS-Will, 2021-07-22:
on your Remote PC, try the command below.
```bash
$ sudo apt remove ros-melodic-turtlebot3-*
$ sudo apt remove ros-melodic-hls-lfcd-lds-driver
$ cd ~/catkin_ws/src
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/turtlebot3.git
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver.git
$ cd ~/catkin_ws && catkin_make && source ~/catkin_ws/devel/setup.bash
```

azhou202, 2021-07-22:
The same issue is still occurring. One that that came up is that when I tried to run `catkin_make`, it would not allow me to do so and instead required me to run `catkin_make_isolated`. I'm not sure if this makes a difference. The full output from terminal when it showed the error is this:

-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/realitylab/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/realitylab/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 13 packages in topological order:
-- ~~  - ros_battleship
-- ~~  - turtlebot3 (metapackage)
-- ~~  - turtlebot3_msgs
-- ~~  - turtlebot3_navigation
-- ~~  - turtlebot3_simulations (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'turtlebot3_simulations' contains non standard content. Use the content of the following file instead: /home/realitylab/catkin_ws/build/catkin_generated/metapackages/turtlebot3_simulations/CMakeLists.txt
-- ~~  - hls_lfcd_lds_driver
-- ~~  - turtlebot3_bringup
-- ~~  - turtlebot3_example
-- ~~  - turtlebot3_fake
-- ~~  - turtlebot3_gazebo
-- ~~  - turtlebot3_slam
-- ~~  - turtlebot3_teleop
-- ~~  - turtlebot3_description
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/realitylab/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/realitylab/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

For context, I have another package inside my src folder in catkin_ws, but it is another ROS package that I wrote, and I don't think it should be interfering with this.

ROBOTIS-Will, 2021-07-23:
Could you try after deleting the `turtlebot3_simulation` directory and clone again?
You might have cloned incorrect source or corrupted the source with other contents.

```bash
$ rm -rf ~/catkin_ws/src/turtlebot3_simulations
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git
$ cd ~/catkin_ws && catkin_make && source ~/catkin_ws/devel/setup.bash
```

azhou202, 2021-07-23:
I tried that and still the same error. I also removed all my other packages from catkin_ws/src and used catkin_make instead of catkin_make_isolated, but still same problem.

ROBOTIS-Will, 2021-07-26:
@azhou202 
Sorry but I can't reproduce the issue you're experiencing.
Did you install the Jetson with the provided image from NVidia?
https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#write
I'd recommend to install packages for TurtleBot3 first and then install other packages for your needs to check which package causes problem.

closed 2024-11-08 by sunghowoo