ROBOTIS-GIT/turtlebot3 #764: Can we use Turtlebot3 package for QBot 2e, a differential drive mobile robot (rasberry pi 3B+ but without openCR)

sridharsidhu0801, 2021-07-26:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ *] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ *] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ *] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [* ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [* ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on the terminal.

    - HERE

9. Please describe the issue in detail.

    I was able to perform SLAM and navigation for the simulated model in Gazebo. Now, I want to implement the turtlebot3 package on the QBot 2e, a differential drive robot. The QBot 2e uses a Raspberry Pi 3 Model B+ as a small-scale embedded computer but it doesn't have the OpenCR board. Can I use the turtlebot3 package on QBot 2e with some modifications without using the OpenCR or do I have to connect the OpenCR board? Please provide me the best suggestion.
    
  Thank you

ROBOTIS-Will, 2021-07-27:
Hi @sridharsidhu0801 

Unfortunately, we do not provide a technical assistant for customizing robots.
Since QBot 2e uses different sensors, a lot of modification will be needed if you are going to use the TurtleBot3 packages.
For example, interfacing the sensors of the QBot 2e to Raspberry Pi could be quite different from TurtleBot3.

Thank you.

sridharsidhu0801, 2021-08-04:
Hello,

Yes, I understand. I am trying to use the packages from the turtlebot2
package as well and trying to build a package using exiting packages for a
particular hardware.

I have a small question, I used the *Raspberry Pi 3B+* ROS1 Melodic
image(Ubuntu 18.04 based)
<https://www.robotis.com/service/download.php?no=2011> image file on to the
robot and trying to connect the robot using ssh by providing the default
password turtlebot (as per the document) but it says permission denied.
Could you help me with this one issue, that will be all?

Thank you

ROBOTIS-Will, 2021-08-06:
Hi @sridharsidhu0801 
There are different reasons for unsuccessful ssh connection.
If you haven't used ssh before, you might be missing the ssh key.
Please google the error message on your terminal and there will be many resolutions.
Thank you.