ROBOTIS-GIT/turtlebot3 #766: May I know how we can obtain transforms of the bot? I want to obtain frame ids, similar to using hector_trajectory_server used in ROS

ROBOTIS-Will, 2021-08-06:
May I know how we can obtain transforms of the bot? I want to obtain frame ids, similar to using hector_trajectory_server used in ROS

_Originally posted by @akshay-30 in https://github.com/ROBOTIS-GIT/turtlebot3/issues/512#issuecomment-890928829_

ROBOTIS-Will, 2021-08-06:
@akshay-30 

You can read a specific frame_id in `/tf` topic.
Some of the the transform of the hardware attached to the robot are published via the `/tf_static` topic.

akshay-30, 2021-08-06:
I tried to read the odom frame and the base_link frame. And publish the frame ids , but I'm getting a error of invalid frame is passed in canTransform : odom frame (target frame) does not exist.

akshay-30, 2021-08-06:
I'm using ubuntu 20.04 version on my laptop, and running ros2 foxy. I have tried to port hector_trajectory_server to ros2. Just want to test it on turtlebot3.

ROBOTIS-Will, 2021-08-09:
While TurtleBot3 is running the Bringup node, you can read the `/tf` and will return both `odom` and `base_link` as below.
[image]

akshay-30, 2021-08-09:
I'm trying to read the frame ids through tf2_ros::TransformListener, passing it into tf2_ros::Buffer. But canTransform and lookupTransform functions are not able to find the target frame ids. I get the error - frame does not exist

akshay-30, 2021-08-09:
Also, turtlebot3_bringup - robot.launch.py isn't working. Process is dying. turtlebot3_node isn't building. 
[image]

akshay-30, 2021-08-10:
Nevermind, I found the bug in my code and was able to debug it with some help. I wasn't passing the right parameters into lookuptransform. Now I am getting the right frame ids and their transforms. 
Thanks for the help @ROBOTIS-Will

ROBOTIS-Will, 2021-08-10:
@akshay-30 
Looks like you haven't correctly installed the package or didn't source the correct working space.
Make sure to install all the necessary packages to run the TurtleBot3.

akshay-30, 2021-08-10:
For installation I followed the tutorial itself. And have sourced it properly. 
turtlebot3_gazebo, turtlebot3_navigation, turtlebot3_teleop are all building properly except turtlebot3_bringup

GyuH13, 2024-10-31:
This issue was raised so long ago that it is now challenging to address. 
Please refer to the updated manual, and if you have any further questions, feel free to open a new issue.

sunghowoo, 2024-11-08:
We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-11-08 by sunghowoo