ROBOTIS-GIT/turtlebot3 #767: Raspberry Pi 4B + ROS1 Melodic： Navigation Error

toukairinn9999, 2021-08-06:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [X] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [X] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [X] Raspberry Pi 4B (2GB or 4GB) ROS1 Melodic image(Raspberry Pi OS based)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [X] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [1.2.2]

7. Specify the commands or instructions to reproduce the issue.

    following the ROBOTIS TurtleBot3 e-Manual

　[Remote PC]
　$ roscore

　[TB3]
　$ roslaunch turtlebot3_bringup turtlebot3_robot.launch

　[Remote PC]
　$ roslaunch turtlebot3_slam turtlebot3_slam.launch

　[Remote PC]
　$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

　I create a map of the room by moving TB3 with the teleop command.

　[Remote PC]
　$ rosrun map_server map_saver -f ~/map
　
　When I finish mapping the room, I terminate the Gmapping and teleop nodes, and move TB3 to the room's reference position by hand.

　[Remote PC]
　roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml

　I use 2D Pose Estimate from Rviz to give the initial position of TB3, and then use 2D Nav Goal to set the navigation goal.

8. Copy and Paste the error messages on terminal.

　When I launch the navigation node, the following error messages on the terminal.

　・[ WARN] [1626940000.681850009]: global_costmap: Parameter "plugins" not provided, loading pre-Hydro parameters
　・[ WARN] [1626940001.125957975]: local_costmap: Parameter "plugins" not provided, loading pre-Hydro parameter

9. Please describe the issue in detail.

At the stage of starting the Navigation node, it is displayed without any problem on Rviz, as shown in the image below.
　
[image]

However, when the goal is given by 2D Nav Goal and TB3 starts moving, the Global Costmap is not updated and TB3's behavior becomes strange.
The image shown below captures the Costmap shifting in three steps from left to right.

[image]

Also, this is the ROS1 Melodic image for Raspberry Pi 4B, which has recently been distributed on e-manual.
　
I think it may be a Melodic-specific problem, because the Kinetic version of TB3 worked fine in the same procedure and under the same actual environment.

I have already tried the following measures.
・Press the reset button on the OpenCR1.0
・Reinstall the ROS package including TB3 package

ROBOTIS-Will, 2021-08-09:
Hi @toukairinn9999 
Thanks for reporting the issue.
I've just tested with the image and looks like the image needs to be updated since it is created in 2019.
According to our test, the Ubuntu 18.04 works very well with RPi4 so we'll create a new Ubuntu 18.04 based TB3 recovery image and upload it in a few days.
Thanks!

ROBOTIS-Will, 2021-08-24:
Hi @toukairinn9999 
My team is currently testing a RPi4 Melodic recovery image.
Once the test is completed, the image will be released via TurtleBot3 eManual.
I'm expecting it to be released by tomorrow at the latest.
Thank you.

sunghowoo, 2024-11-08:
We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-11-08 by sunghowoo