ROBOTIS-GIT/turtlebot3 #768: 启动turtlebot3-burger ，执行：roslaunch turtlebot3_bringup turtlebot3_robot.launch  报错  ：Creation of publisher failed: Checksum does not match

gao89704, 2021-08-09:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ √] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [√] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [√] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ √] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [√] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.15.11]
    - Firmware version: [1.4.18]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - gst@gst-desktop:~$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/gst/.ros/log/55dba076-f903-11eb-ad27-2d4d313a22b1/roslaunch-gst-desktop-1517.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.115:41617/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

auto-starting new master
process[master]: started with pid [1527]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 55dba076-f903-11eb-ad27-2d4d313a22b1
process[rosout-1]: started with pid [1537]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [1540]
process[turtlebot3_lds-3]: started with pid [1541]
process[turtlebot3_diagnostics-4]: started with pid [1542]
[INFO] [1628507833.529364]: ROS Serial Python Node
[INFO] [1628507833.545574]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1628507835.659724]: Requesting topics...
[INFO] [1628507835.691602]: Note: publish buffer size is 1024 bytes
[INFO] [1628507835.695826]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1628507835.705905]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1628507835.889515]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1628507835.900551]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1628507835.970381]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1628507835.981292]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1628507835.988779]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1628507835.999517]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1628507836.178900]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1628507836.192451]: Note: subscribe buffer size is 1024 bytes
[INFO] [1628507836.197093]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1628507836.209833]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1628507836.223561]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1628507836.236959]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1628507838.801621]: Setup TF on Odometry [odom]
[INFO] [1628507838.808961]: Setup TF on IMU [imu_link]
[INFO] [1628507838.815788]: Setup TF on MagneticField [mag_link]
[INFO] [1628507838.822422]: Setup TF on JointState [base_link]
[ERROR] [1628507838.830881]: Tried to publish before configured, topic id 131
[INFO] [1628507838.835459]: Requesting topics...
[INFO] [1628507841.362751]: Calibration End
[INFO] [1628507841.374199]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1628507841.381593]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1628507841.388980]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1628507841.396946]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1628507841.404479]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1628507841.411742]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1628507841.418846]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1628507841.426322]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1628507841.433626]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1628507841.449721]: Tried to publish before configured, topic id 131
[INFO] [1628507841.454558]: Requesting topics...
[ERROR] [1628507841.545956]: Tried to publish before configured, topic id 131
[INFO] [1628507841.550947]: Requesting topics...
[ERROR] [1628507841.591684]: Tried to publish before configured, topic id 131
[INFO] [1628507841.596433]: Requesting topics...
[ERROR] [1628507841.637244]: Tried to publish before configured, topic id 131
[INFO] [1628507841.641865]: Requesting topics...
[ERROR] [1628507841.677866]: Tried to publish before configured, topic id 131
[INFO] [1628507841.682669]: Requesting topics...

9. Please describe the issue in detail.

    - HERE

ROBOTIS-Will, 2021-08-10:
Hi @gao89704 
You need to upload the Noetic OpenCR firmware.
Please find the OpenCR setup section of the eManual and make sure to select `Noetic` on top of the page.
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup

Leo5017, 2023-03-29:
[INFO] [1679890453.957172]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1679890453.958169]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1679890453.960164]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1679890453.962158]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1679890453.963156]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1679890453.965151]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1679890453.966148]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1679890453.969140]: Cannot import package : turtlebot3_msgs
[ERROR] [1679890453.970138]: sys.path was ['C:\\opt\\ros\\noetic\\x64\\python38.zip', 'C:\\opt\\ros\\noetic\\x64', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages', 'c:\\opt\\ros\\noetic\\x64\\lib/site-packages', 'c:\\opt\\ros\\noetic\\x64\\Lib\\site-packages', 'C:\\opt\\ros\\noetic\\x64\\Lib\\rosserial_python', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32\\lib', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\Pythonwin']
[ERROR] [1679890453.971255]: Creation of subscriber failed: 'NoneType' object has no attribute 'msg'
[ERROR] [1679890453.973712]: Tried to publish before configured, topic id 125
[INFO] [1679890453.974677]: Requesting topics...
[ERROR] [1679890453.980886]: Tried to publish before configured, topic id 125
[INFO] [1679890453.981972]: Requesting topics...
[ERROR] [1679890453.993539]: Cannot import package : turtlebot3_msgs
[ERROR] [1679890453.995499]: sys.path was ['C:\\opt\\ros\\noetic\\x64\\python38.zip', 'C:\\opt\\ros\\noetic\\x64', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages', 'c:\\opt\\ros\\noetic\\x64\\lib/site-packages', 'c:\\opt\\ros\\noetic\\x64\\Lib\\site-packages', 'C:\\opt\\ros\\noetic\\x64\\Lib\\rosserial_python', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32\\lib', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\Pythonwin']
[ERROR] [1679890453.996497]: Creation of publisher failed: 'NoneType' object has no attribute 'msg'
[ERROR] [1679890453.999520]: Cannot import package : turtlebot3_msgs
[ERROR] [1679890454.000519]: sys.path was ['C:\\opt\\ros\\noetic\\x64\\python38.zip', 'C:\\opt\\ros\\noetic\\x64', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages', 'c:\\opt\\ros\\noetic\\x64\\lib/site-packages', 'c:\\opt\\ros\\noetic\\x64\\Lib\\site-packages', 'C:\\opt\\ros\\noetic\\x64\\Lib\\rosserial_python', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32\\lib', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\Pythonwin']
[ERROR] [1679890454.001516]: Creation of publisher failed: 'NoneType' object has no attribute 'msg'
[INFO] [1679890454.002356]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1679890454.004353]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1679890454.006380]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1679890454.008353]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1679890454.009339]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1679890454.011017]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1679890454.012013]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1679890454.015947]: Cannot import package : turtlebot3_msgs
[ERROR] [1679890454.016945]: sys.path was ['C:\\opt\\ros\\noetic\\x64\\python38.zip', 'C:\\opt\\ros\\noetic\\x64', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages', 'c:\\opt\\ros\\noetic\\x64\\lib/site-packages', 'c:\\opt\\ros\\noetic\\x64\\Lib\\site-packages', 'C:\\opt\\ros\\noetic\\x64\\Lib\\rosserial_python', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32\\lib', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\Pythonwin']
[ERROR] [1679890454.017943]: Creation of subscriber failed: 'NoneType' object has no attribute 'msg'
[ERROR] [1679890454.021964]: Cannot import package : turtlebot3_msgs
[ERROR] [1679890454.022964]: sys.path was ['C:\\opt\\ros\\noetic\\x64\\python38.zip', 'C:\\opt\\ros\\noetic\\x64', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages', 'c:\\opt\\ros\\noetic\\x64\\lib/site-packages', 'c:\\opt\\ros\\noetic\\x64\\Lib\\site-packages', 'C:\\opt\\ros\\noetic\\x64\\Lib\\rosserial_python', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32\\lib', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\Pythonwin']
[ERROR] [1679890454.024115]: Creation of publisher failed: 'NoneType' object has no attribute 'msg'
[ERROR] [1679890454.026529]: Cannot import package : turtlebot3_msgs
[ERROR] [1679890454.027494]: sys.path was ['C:\\opt\\ros\\noetic\\x64\\python38.zip', 'C:\\opt\\ros\\noetic\\x64', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages', 'c:\\opt\\ros\\noetic\\x64\\lib/site-packages', 'c:\\opt\\ros\\noetic\\x64\\Lib\\site-packages', 'C:\\opt\\ros\\noetic\\x64\\Lib\\rosserial_python', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\win32\\lib', 'C:\\opt\\ros\\noetic\\x64\\lib\\site-packages\\Pythonwin']
[ERROR] [1679890454.028492]: Creation of publisher failed: 'NoneType' object has no attribute 'msg'
[... cut ...]
[INFO] [1679890454.033638]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1679890454.034636]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1679890454.036630]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1679890454.039622]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1679890454.041650]: Setup publisher on /tf [tf/tfMessage]

Sorry,I have a question.How do I solve it?

Turtlebots3 waffle Pi
RPi4
Windows

sunghowoo, 2024-11-08:
For ROS1 Noetic, the correct model designation is waffle_noetic or buger_noetic. By updating the environment variable accordingly, the issue was resolved. . If this issue recurs, please feel free to submit it again.

closed 2024-11-08 by sunghowoo