ROBOTIS-GIT/turtlebot3 #769: turtlebot OpenCR + Jetson Nano Configuration - Dynamixel XM430 not spinnning

PraveenElango, 2021-08-10:
I swapped out the Raspberry Pi in the turtlebot for a Jetson Nano and followed the OpenCR setup steps outlined here: https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/

I was able to upload the firmware successfully to the OpenCR (from both the Jetson and from my PC using the Arduino IDE). However, when I pressed the switches for the OpenCR test to observe if the Dynamixel motors were spinning upon pressing the switch buttons, they didn’t spin at all.

I then did the Dynamixel setup for the OpenCR (https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3) and the motors spinned when I uploaded the test motor code to the board using the arduino IDE from my PC. I then uploaded the turtlebot3_core to my board and connected the OpenCR to my Jetson. However, the motors still didn’t spin after I pressed the buttons (the buttons are functional as I did another test using the IDE from the PC). I am not sure how to rectify this, and I could do with some help

ROBOTIS-Will, 2021-08-10:
Hi @PraveenElango 
If you have uploaded ROS2 Arduino example on OpenCR via Arduino IDE, it may not work correctly as there are some issues with the current source code.
Please try uploading the ROS2 OpenCR firmware with the script described in the eManual
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup-4
Thank you.

PraveenElango, 2021-08-10:
Hi @ROBOTIS-Will , thanks for getting back! I am using ROS Melodic on the Jetson and I did upload the ROS1 OpenCR firmware with the script described in the eManual on the OpenCR. The command line did return a successful firmware upload, but when I tried pressing the buttons for the OpenCR test, the Dynamixels didn't spin.

ROBOTIS-Will, 2021-08-10:
@PraveenElango 
When testing the DYNAMIXEL with uploaded firmware, you may need to press the button a bit longer, like 2~3 seconds.
A brief click of the button won't activate the testing process.
Thanks.

PraveenElango, 2021-08-10:
@ROBOTIS-Will You're right, it works now! Thanks for the help :)

PraveenElango, 2021-08-11:
Hi, if uploading the firmware using Arduino IDE would not work since there are issues with the source code, how would I be able to upload the turtlebot3_core (https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino)  on the OpenCR for a ROS implementation?

ROBOTIS-Will, 2021-08-12:
@PraveenElango 
Please refer to the eManual below.
Make sure to select the correct ROS version at the top of the page.
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup

PraveenElango, 2021-08-16:
@ROBOTIS-Will Apologies for the late reply, and thanks for the help - it works now! One more question out of sheer curiosity - if I wanted to include custom ROS publishers and subscribers, would it be possible for me to do this by editing the firmware files like waffle.opencr? If I am still using ROS1, is using the Arduino IDE to upload the firmware on the OpenCR still an option? Thanks!

ROBOTIS-Will, 2021-08-17:
Hi @PraveenElango 
No worries.
Yes, ROS1 OpenCR firmware can be uploaded via Arduino IDE.
In case you need to attach additional sensors to OpenCR, you may revise the firmware and upload it to OpenCR.
Thanks!

PraveenElango, 2021-08-20:
Great, thank you!

closed 2021-08-20 by PraveenElango