ROBOTIS-GIT/turtlebot3 #772: Turtlebot3 waffle pi [Noetic]- OpenCR Test Failure

nullbyte91, 2021-08-14:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ *] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ *] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ *] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ *] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ *] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.
    **OpenCR Setup:[RPI]**

```bash
    export OPENCR_PORT=/dev/ttyACM0
    export OPENCR_MODEL=waffle_noetic
    rm -rf ./opencr_update.tar.bz2 && wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2
    tar -xjf ./opencr_update.tar.bz2 && cd ~/opencr_update && ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr
```
Only one actuator was working when I tried pressing SW2 button from OpenCR HW. 
Then I tried Setup DYNAMIXELs using Ardunino by following below link, https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3

Now I'm getting the below error from Ardunino serial port when i tried to setup motor,
```bash
1. setup left  motor
2. setup right motor
3. test  left  motor
4. test  right motor
>> Do you really want to setup ? y/n : yes
setup.... left
Find Motor...
    ... SUCCESS
    [ID: 1 found at baud: 1000000]
Setup Motor Left...
[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!
[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!
[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!
[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!
[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!
    ok
```

nullbyte91, 2021-08-17:
@ROBOTIS-Will Any suggestions?

ROBOTIS-Will, 2021-08-17:
@nullbyte91 
During the motor setup, make sure to supply 12V power to OpenCR and turn on the switch, and connect only one DYNAMIXEL at a time.
Thank you.

nullbyte91, 2021-08-17:
@ROBOTIS-Will  Thank you so much, It worked.

closed 2021-08-17 by nullbyte91