ROBOTIS-GIT/turtlebot3 #773: Turtlebot Don't Move When I Publish To cmd_vel

Bo-Ru, 2021-08-15:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [X] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [X] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

Compiled and ran the following code:
```
#include <memory>
#include <chrono>
#include "rclcpp/rclcpp.hpp"
#include "geometry_msgs/msg/twist.hpp"
using namespace std::chrono_literals;
class MinimalPublisher : public rclcpp::Node
{
public:
        MinimalPublisher() : Node("pub_node")
        {   
                publisher_ = this->create_publisher<geometry_msgs::msg::Twist>("cmd_vel", 1); 
                timer_ = this->create_wall_timer(250ms, std::bind(&MinimalPublisher::publish_message, this));
        }   
private:
        void publish_message()
        {   
                geometry_msgs::msg::Twist cmd_vel;
                cmd_vel.linear.x = 10.0;
                RCLCPP_INFO(this->get_logger(), "Sending Linear: '%f'", cmd_vel.linear.x);
                publisher_->publish(cmd_vel);
        }   
        rclcpp::TimerBase::SharedPtr timer_;
        rclcpp::Publisher<geometry_msgs::msg::Twist>::SharedPtr publisher_;
};
int main(int argc, char * argv[])
{
        rclcpp::init(argc, argv);
        rclcpp::spin(std::make_shared<MinimalPublisher>());
        rclcpp::shutdown();
        return 0;
}
```

8. Copy and Paste the error messages on terminal.

Recently, every time I start Bringup, it shows the following message:
```
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
```

9. Please describe the issue in detail.

The raspberry pi on Turtlebot seems to be subscribing cmd_vel as it should be, and running `ros2 topic info cmd_vel` with the code running and bringup running also shows one publisher and one subscriber. Using `ros2 topic echo cmd_vel` echos the message I published no problem, but still, Turtlebot is not moving. However, when I run keyboard teleop, which is written in python for foxy, works no problem. Turtlebot is moving normally, but I don't know why Turtlebot is not moving when running the code since the format the published message is is all the same.

ashekim, 2021-08-17:
Hi @Bo-Ru .
BURGER_MAX_LIN_VEL is 0.22 ([`link`](https://github.com/ROBOTIS-GIT/turtlebot3/blob/f5e572a5a7dd0d8d6f76b23848f8a0b9411b489f/turtlebot3_teleop/turtlebot3_teleop/script/teleop_keyboard.py#L51))
Set cmd_vel.linear.x to less than 0.22.

Thank you.

ROBOTIS-Will, 2021-08-17:
Hi @Bo-Ru 

Regarding the `UserWarning`, this is a deprecation warning and is not critical in operating the robot.
At the moment, please clone the source and build from it if you don't want to see them.
https://github.com/ros/rosdistro/pull/29243
Will fix this issue soon.

Thanks!

Bo-Ru, 2021-08-17:
👍Thanks, changing the linear value below 0.22 worked!

closed 2021-08-17 by Bo-Ru