ROBOTIS-GIT/turtlebot3 #777: RVIZ Issue

binaryhellstorm, 2021-08-28:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [X] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [X] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [X] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.12.14]
    - Firmware version: [1.2.2]

7. Specify the commands or instructions to reproduce the issue.

Normal bringup

8. Copy and Paste the error messages on terminal.

user@ROSCore1:~$ roslaunch turtlebot3_slam turtlebot3_slam.launch
... logging to /home/user/.ros/log/b08239c6-083b-11ec-9cd4-b8ca3a9c216b/roslaunch-ROSCore1-6243.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.2.202:41083/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.17
 * /turtlebot3_slam_gmapping/angularUpdate: 0.2
 * /turtlebot3_slam_gmapping/astep: 0.05
 * /turtlebot3_slam_gmapping/base_frame: base_footprint
 * /turtlebot3_slam_gmapping/delta: 0.05
 * /turtlebot3_slam_gmapping/iterations: 5
 * /turtlebot3_slam_gmapping/kernelSize: 1
 * /turtlebot3_slam_gmapping/lasamplerange: 0.005
 * /turtlebot3_slam_gmapping/lasamplestep: 0.005
 * /turtlebot3_slam_gmapping/linearUpdate: 1.0
 * /turtlebot3_slam_gmapping/llsamplerange: 0.01
 * /turtlebot3_slam_gmapping/llsamplestep: 0.01
 * /turtlebot3_slam_gmapping/lsigma: 0.075
 * /turtlebot3_slam_gmapping/lskip: 0
 * /turtlebot3_slam_gmapping/lstep: 0.05
 * /turtlebot3_slam_gmapping/map_frame: map
 * /turtlebot3_slam_gmapping/map_update_interval: 2.0
 * /turtlebot3_slam_gmapping/maxUrange: 3.0
 * /turtlebot3_slam_gmapping/minimumScore: 50
 * /turtlebot3_slam_gmapping/odom_frame: odom
 * /turtlebot3_slam_gmapping/ogain: 3.0
 * /turtlebot3_slam_gmapping/particles: 100
 * /turtlebot3_slam_gmapping/resampleThreshold: 0.5
 * /turtlebot3_slam_gmapping/sigma: 0.05
 * /turtlebot3_slam_gmapping/srr: 0.1
 * /turtlebot3_slam_gmapping/srt: 0.2
 * /turtlebot3_slam_gmapping/str: 0.1
 * /turtlebot3_slam_gmapping/stt: 0.2
 * /turtlebot3_slam_gmapping/temporalUpdate: 0.5
 * /turtlebot3_slam_gmapping/xmax: 10.0
 * /turtlebot3_slam_gmapping/xmin: -10.0
 * /turtlebot3_slam_gmapping/ymax: 10.0
 * /turtlebot3_slam_gmapping/ymin: -10.0

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    turtlebot3_slam_gmapping (gmapping/slam_gmapping)

ROS_MASTER_URI=http://192.168.2.202:11311

process[robot_state_publisher-1]: started with pid [6263]
process[turtlebot3_slam_gmapping-2]: started with pid [6264]
process[rviz-3]: started with pid [6265]
QXcbConnection: Could not connect to display 
================================================================================REQUIRED process [rviz-3] has died!
process has died [pid 6265, exit code -6, cmd /opt/ros/kinetic/lib/rviz/rviz -d /opt/ros/kinetic/share/turtlebot3_slam/rviz/turtlebot3_gmapping.rviz __name:=rviz __log:=/home/user/.ros/log/b08239c6-083b-11ec-9cd4-b8ca3a9c216b/rviz-3.log].
log file: /home/user/.ros/log/b08239c6-083b-11ec-9cd4-b8ca3a9c216b/rviz-3*.log
Initiating shutdown!
================================================================================
[rviz-3] killing on exit
[turtlebot3_slam_gmapping-2] killing on exit
[robot_state_publisher-1] killing on exit
virtual GMapping::GridSlamProcessor::~GridSlamProcessor(): Start
virtual GMapping::GridSlamProcessor::~GridSlamProcessor(): Deleting tree
shutting down processing monitor...
... shutting down processing monitor complete
done

9. Please describe the issue in detail.

I am having an issue where I am unable to bring up RVIZ. I have completely reloaded the PI from the Robotis SD card image and followed the set up guide, I also reloaded the "remote" PC with a fresh Ubuntu 16.04 install and the issue persists.  Teleoperation still works so I know the two systems are able to communicate with each other.

ROBOTIS-Will, 2021-08-30:
Hi @binaryhellstorm 
Have you correctly installed Ubuntu 16.04 LTS Desktop image on your Remote PC?
Installing the Ubuntu server may require additional setup for GUI environment.
If you are remotely connecting to the Remote PC, I'd recommend using a remote desktop.
Thanks.

sunghowoo, 2024-11-08:
We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-11-08 by sunghowoo