ROBOTIS-GIT/turtlebot3 #779: Tf errors in turtlebot3 simulation

abdur78, 2021-09-02:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [* ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [* ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ *] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    I am following the simulation tutorials from the emanual: https://emanual.robotis.com/docs/en/platform/turtlebot3/slam_simulation/.
All the packages have been setup as described in the manual. First I run the gazebo world launch file, and then the slam gmapping node. 

8. Copy and Paste the error messages on terminal.

    The gazebo node is launched perfectly:

process[gazebo-1]: started with pid [43183]
process[gazebo_gui-2]: started with pid [43188]
process[spawn_urdf-3]: started with pid [43193]
[ INFO] [1630574215.657683157]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1630574215.658913882]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1630574215.781753867]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1630574215.783198073]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1630574216.071391078]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1630574216.099642564, 0.008000000]: Physics dynamic reconfigure ready.
[spawn_urdf-3] process has finished cleanly
log file: /home/syed/.ros/log/6550f358-0bcd-11ec-9c93-a729c51044a2/spawn_urdf-3*.log

When I run the slam gmapping node: the following warning occurs:

process[robot_state_publisher-1]: started with pid [43875]
process[turtlebot3_slam_gmapping-2]: started with pid [43876]
process[rviz-3]: started with pid [43877]
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom at time 100.401000 according to authority unknown_publisher
         at line 278 in /home/syed/catkin_ws/src/geometry2/tf2/src/buffer_core.cpp

I have also tried with hector slam and cartographer. There is no warning or with hector slam. But with cartographer, the following errors occur:

process[robot_state_publisher-1]: started with pid [29547]
process[move_base-2]: started with pid [29548]
process[cartographer_node-3]: started with pid [29549]
process[cartographer_occupancy_grid_node-4]: started with pid [29550]
process[flat_world_imu_node-5]: started with pid [29556]
process[rviz-6]: started with pid [29561]
[ INFO] [1630570355.044080255]: I0902 10:12:35.000000 29549 configuration_file_resolver.cc:41] Found '/opt/ros/noetic/share/turtlebot3_slam/config/turtlebot3_lds_2d.lua' for 'turtlebot3_lds_2d.lua'.
[ INFO] [1630570355.046582780]: I0902 10:12:35.000000 29549 configuration_file_resolver.cc:41] Found '/home/syed/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1630570355.046680970]: I0902 10:12:35.000000 29549 configuration_file_resolver.cc:41] Found '/home/syed/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1630570355.046822276]: I0902 10:12:35.000000 29549 configuration_file_resolver.cc:41] Found '/home/syed/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1630570355.046879433]: I0902 10:12:35.000000 29549 configuration_file_resolver.cc:41] Found '/home/syed/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1630570355.047377654]: I0902 10:12:35.000000 29549 configuration_file_resolver.cc:41] Found '/home/syed/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1630570355.047581418]: I0902 10:12:35.000000 29549 configuration_file_resolver.cc:41] Found '/home/syed/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1630570355.047843333]: I0902 10:12:35.000000 29549 configuration_file_resolver.cc:41] Found '/home/syed/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1630570355.047923700]: I0902 10:12:35.000000 29549 configuration_file_resolver.cc:41] Found '/home/syed/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1630570355.048212937]: I0902 10:12:35.000000 29549 configuration_file_resolver.cc:41] Found '/home/syed/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1630570355.048357948]: I0902 10:12:35.000000 29549 configuration_file_resolver.cc:41] Found '/home/syed/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1630570355.090280983]: I0902 10:12:35.000000 29549 map_builder_bridge.cc:135] Added trajectory with ID '0'.
[ WARN] [1630570358.129078186, 138.712000000]: W0902 10:12:38.000000 29549 node.cc:881] Expected topic "scan" (trajectory 0) (resolved topic "/scan") but no publisher is currently active.
[ WARN] [1630570358.129378635, 138.712000000]: W0902 10:12:38.000000 29549 node.cc:881] Expected topic "odom" (trajectory 0) (resolved topic "/odom") but no publisher is currently active.
[ WARN] [1630570358.129493393, 138.712000000]: W0902 10:12:38.000000 29549 node.cc:890] Currently available topics are: /rosout_agg,/rosout,/clock,/gazebo/link_states,/gazebo/model_states,/gazebo/performance_metrics,/gazebo/parameter_descriptions,/gazebo/parameter_updates,/tf,/tf_static,/flat_imu,/map,/submap_list,/trajectory_node_list,/landmark_poses_list,/constraint_list,/scan_matched_points2,/cmd_vel,/move_base/current_goal,/move_base/goal,/move_base/recovery_status,/initialpose,/move_base_simple/goal,
[ WARN] [1630570360.580602108, 141.003000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.
[ WARN] [1630570366.043921127, 146.030000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.101 timeout was 0.1.

9. Please describe the issue in detail.

When I run the slam node, there is an error on rviz that there are no transforms from odom to base. The tf tree also shows that there is no link between odom and base_footprint. Odom and map are connected to each other but are separate from the rest of tree.: 
[image]

No matter which slam method I try (hector, cartographer or gmapping), the error remains the same on rviz:
[image]

The info of robot state publisher is:
Node [/robot_state_publisher]
Publications: 
 * /rosout [rosgraph_msgs/Log]
 * /tf [tf2_msgs/TFMessage]
 * /tf_static [tf2_msgs/TFMessage]

Subscriptions: 
 * /clock [rosgraph_msgs/Clock]
 * /joint_states [unknown type]

Services: 
 * /robot_state_publisher/get_loggers
 * /robot_state_publisher/set_logger_level

Also, when I run the teleportation node, the robot does not move in gazebo. I have been stuck here for quite a while now, and would really appreciate the help. Thanks.

ashekim, 2021-09-06:
Hi @abdur78 . Thank you for your questions.

Did you install the necessary packages on the Remote PC before running the simulation?
When using gazebo, you need to install 3.1.3 Install [Dependent ROS Packages](https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#install-dependent-ros-packages-2)

Thank you.

DingoOz, 2021-10-04:
For each terminal are you remembering to go into your catkin_ws directory and "source devel/setup.bash" before running other commands?

15910693796, 2022-03-07:
You can try to start each child launch file separately without directly starting the total launch file. It's amazing: I started the turtlebot3_navigation.launch in ROS kinetic to display TF error, but started each child launch file separately without error.

sunghowoo, 2024-12-05:
Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. This issue appears to be resolved, so it will be closed.

closed 2024-12-05 by sunghowoo