ROBOTIS-GIT/turtlebot3 #781: Dynamixels not moving to teleop keyboard

BrandonTech, 2021-09-07:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    -  Burger

2. Which ROS is working with TurtleBot3?

    - ROS 1 Melodic Morenia

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - Raspberry Pi 3B+

4. Which OS you installed on SBC?

    -  Raspbian distributed by ROBOTIS

5. Which OS you installed on Remote PC?

    -  Ubuntu 18.04 LTS (Bionic Beaver)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.14.10] - ROS
    - Firmware version: [1.2.6] - OpenCR Core

7. Specify the commands or instructions to reproduce the issue.

    - *(export TURTLEBOT3_MODEL=burger is predefined in bashrc file)
    - roscore on remote PC (success)
    - opened new terminal and ssh into turtlebot (success) 
    - called roslaunch turtlebot3_bringup turtlebot3_robot.launch from turtlebot (success) 
    - opened another terminal and called roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch from remote pc (success) 
    - pressed w,a,d,x while in focus of that terminal and dynamixels did not move
    

8. Copy and Paste the error messages on terminal.

    - N/A

9. Please describe the issue in detail.

    - Everything seems to be working up until it's time to control the dynamixels via keyboard teleoperation. I read through some similar issues (open and closed) and nothing works for me. I have "source /opt/ros/melodic/setup.bash" defined in my bashrc file. The motors do move when I push both SW1 and SW2 buttons. I setup the motors multiples times following the "Setup DYNAMIXELs for TurtleBot3" section of the eManual. I see the values changing on the rostopic monitor and in the terminal from the remote pc with rostopic echo /cmd_vel but no values when I call that from the turtebot. I have attached a few screenshots just in case they can be of help. Sorry for zooming out so much on the bringup terminal.
    - 
[image]
[image]
[image]
[image]

ROBOTIS-Will, 2021-09-07:
Hi @BrandonTech 
Please check your ROS network configuration in the `.bashrc` file.
You should use the same address for ROS_MASTER_URI.

- Remote PC : https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#network-configuration
- SBC
[image]

BrandonTech, 2021-09-08:
Hey Will! It seems to be correct. On my remote pc, I have 
export ROS_MASTER_URI=http://172.20.10.3:11311
export ROS_HOTSTNAME=172.20.10.3

and on the Raspberry Pi, I have:
export ROS_MASTER_URI=http://172.20.10.3:11311
export ROS_HOTSTNAME=172.20.10.4

ashekim, 2021-09-10:
Hi. @BrandonTech  

When run roscore, the launch server shows up as http://brandon-Macbookair .
I think it should be modified with your remote pc ip.

- If you look in your roscore terminal window you can see that the launch server is http://brandon-Macbookair
[image]

And I think it would be good to check it using rqt.
Thank you.

BrandonTech, 2021-09-15:
HI @ashekim, I'm not sure why it shows the name of my pc but it is set to the IP of the remote pc
[image]

ashekim, 2021-09-15:
Hi @BrandonTech 
enter the command below
`$rqt`
Please check the Node Graph

[image]

Additionally, the [link](https://wiki.ros.org/ROS/NetworkSetup#Single_machine_configuration) may be helpful.
Thank you.

BrandonTech, 2021-09-16:
Hey @ashekim Thank you for the help with the node graph and link! After seeing the keyboard wasn't connected on the graph, I ended up doing echo $ROS_HOSTNAME on both PC & Turtlebot and saw it was blank on the pc but not turtlebot. After looking very closely in my bashrc, I saw there was a typo that I never realized because my screen is small and my brain read the right word every single time lol. So now it's working perfectly! Thank you again, I shall close this issue now.

closed 2021-09-16 by BrandonTech