ROBOTIS-GIT/turtlebot3 #782: Unable to read position and velocity of dynamixel motors in ROS

amaheshwari2509, 2021-09-13:
Hello,

I am providing the actuation to three motors to my robotics system (custom changed of Waffle PI) from publisher node (designed in ROS) using /cmd_vel topic over twist message. This part is successfully achieved. Further, I am NOT able  to read the "present position and velocity (Control Table Address : 128 & 132)" which is to be used as a sensor feedback for my control algorithm. My goals is to receive these values (from Motor Driver.cpp) and pass onto custom-designed Publisher node (turtlebot3_core.ino). While uploading my code via Arduino IDE, I am not receiving any compiler error  and upload is successful. **I am writing position to third motor and writing velocity to first and second motor. I want to read  position and velocity of all the three motors.**

Note : I am able to read the values successfully via Matlab-Dynamixel SDK and Workbench. 

Step 1 : I have written read position and velocity method based on control table and on same semantics of read encoder.
[image]

Step 2 and 3: Made a topic (with Twist message type) and passing values from read position and velocity (step 1) in this message. Further calling the method in loop. (Upload successful with no compiler errors)

[image]

Step 4 (following image): I am receiving zero values of position and velocity  ("else condition" defined in step 3) . I believe it means methods defined in motor driver unable to read dynamixel motor (dxl_comm_result_pos and dxl_comm_result_vel are false even after calling read position and read velocity methods). It is given by left image of terminal snapshot displaying all zero values. 

I asked this query to know how to received desired output (in the position and velocity ranges defined by control table) shown in right side of following image. 

[image]

Currently I am doing from MATLAB Dynamixel SDK sucesfully ( receiving position and velocity in desired ranges as per the measured value) but I wish to do it from ROS (C++). Looking forward to your response. 

Thanks
Best Regards

ashekim, 2021-09-15:
Hello @amaheshwari2509 . Thanks for the question.

Regarding this inquiry, I would like to know what kind of environment it is.
We would appreciate it if you could inform us about the os version and ros version according to the format below.

1. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

3. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

4. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

Thank you.

amaheshwari2509, 2021-09-16:
Hello,

Thanks for your reply. Following are the answers to your questions :

1 Which ROS is working with TurtleBot3?

 ROS 1 Kinetic

2 Which SBC(Single Board Computer) is working on TurtleBot3?

Raspberry PI 3

3 Which OS you installed on SBC?

Raspbian

4 Which OS you installed on Remote PC?

Ubuntu 16.04 LTS (Xenial Xerus)

amaheshwari2509, 2021-09-25:
> Hello @amaheshwari2509 . Thanks for the question.
>
> Regarding this inquiry, I would like to know what kind of environment it is.
> We would appreciate it if you could inform us about the os version and [...]

Hello @ashekim , 

Hope you doing great. I would like to know were you able to find an issue in my code scripts to read the present position of the motors?

Kind Regards

ashekim, 2021-09-30:
Hi @amaheshwari2509. I'm sorry for the late reply.
Please understand that we do not provide a thorough support on customizing the source code.

About your inquiry we are trying to read the motor's position. but we'd like to apologize for taking the time to check the problem.

Thank you.

amaheshwari2509, 2021-09-30:
> Hi @amaheshwari2509. I'm sorry for the late reply.
> 
> Please understand that we do not provide a thorough support on customizing the source code.
> 
> 
> 
> About your inquiry we are trying to read the motor's position. but we'd like to apologize for taking the time to check the problem.
> 
> 
> 
> Thank you.

Sure @ashekim thank you for putting your valuable time into the issue. I understand and acknowledge the same. Looking forward to know about in near future. 

Kind Regards

closed 2024-11-08 by sunghowoo