ROBOTIS-GIT/turtlebot3 #783: SBC image not downloading

thehelpfulcosmopolitan, 2021-09-17:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [Y] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [Y] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [Y ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Not yet installed)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [Y] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - I am trying to download the SBC image for installing on my SD card. But, it is not downloading on Ubuntu or Windows 10.

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    -  I am trying to download the SBC image for installing on my SD card. But, it is not downloading on Ubuntu or Windows 10. I am getting the error - "Failed - Network error" after it goes all the way to 1.9 GB.

ROBOTIS-Will, 2021-09-27:
Hi @thehelpfulcosmopolitan 

I'm sorry about the delayed response.
Unfortunately, it is difficult for us to identify the cause of the network failure as this is the only case I encountered.
Please use the link below to download the ROS1 Noetic image for Raspberry Pi 3B+.
The link will be available for the next two weeks so make sure to keep a copy of it.
https://drive.google.com/file/d/1FA1ih_rhSihQcLnh8KORWS4i_ReBWq7X/view?usp=sharing

Thank you.

ROBOTIS-Shibata, 2021-09-27:
For your reference. DL is possible from Japan without problem. I don't think it's a problem with ROBOTIS download page.
[image]
Best regards,

closed 2024-11-08 by sunghowoo