ROBOTIS-GIT/turtlebot3 #784: Error building turtlebot3 repositories

denesh-globotix, 2021-09-18:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ X] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ X] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: []
    - Firmware version: []

7. Specify the commands or instructions to reproduce the issue.

    - colcon build

8. Copy and Paste the error messages on terminal.

/home/user/dev_ws/src/turtlebot3/turtlebot3_node/src/odometry.cpp: In constructor ‘robotis::turtlebot3::Odometry::Odometry(std::shared_ptr<rclcpp::Node>&, double, double)’:
/home/user/dev_ws/src/turtlebot3/turtlebot3_node/src/odometry.cpp:39:58: error: no matching function for call to ‘rclcpp::Node::declare_parameter<std::string>(const char [18])’
   39 |   nh_->declare_parameter<std::string>("odometry.frame_id");

9. Please describe the issue in detail.

    - When trying to build the turtlebot3 repository, the above error appeared

ROBOTIS-Will, 2021-09-27:
Hi @denesh-globotix 

When building from the source, you should use a specific branch for example `foxy-devel` for Foxy.
`ros2` and `ros2-devel` are reserved for ROS2 Rolling update for the upcoming ROS2 distribution.
Thank you.

sunghowoo, 2024-11-08:
We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-11-08 by sunghowoo