ROBOTIS-GIT/turtlebot3 #785: How to change the wheels of TurtleBot3 to crawlers by adding two dynamixel motors

Yoshihito-M, 2021-09-21:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ yes] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [yes ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [yes ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [yes ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ yes] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [ROS Kinetic]
    - Firmware version: [x.]

7. Specify the commands or instructions to reproduce the issue.

    - None

8. Copy and Paste the error messages on terminal.

    - None

9. Please describe the issue in detail.

Hi,everyone. I am changing the wheels of TurtleBot3 to crawlers by adding two dynamixel motors. Referring to https://emanual.robotis.com/docs/en/platform/turtlebot3/locomotion/#friendslocomotion, I could assemble caterpillar units as hardware. However, I have question what files I should fix. A while ago, I asked the question to Japan branch office of ROBOTIS, but the branch manager said we could not answer the question because of technical staff had left the office.
I would appreciate it if you could help me fix the program files.

Thanks for your help.

ashekim, 2021-09-30:
Hello @Yoshihito-M. Thank you for your questions.
Please understand that we do not provide a thorough support on customizing the source code.

It is similar to #782, so please refer to it.
Also, if this problem is solved, I think your problem will be solved as well.
Thank you.

ROBOTIS-Will, 2021-09-30:
Hi @Yoshihito-M 
The TurtleBot3 Friends are running without ROS to show various types of locomotion.
If you are trying to add a secondary wheel on each side, you may add additional DYNAMIXEL in the OpenCR TurtleBot3 firmware so that both wheels on each side rotates at the same speed.
However, please be aware when physically connecting both wheels with the caterpillar may introduce an issue caused by a slight discrepancy in speed between the connected wheels.
Thank you.

closed 2022-10-18 by Yoshihito-M