ROBOTIS-GIT/turtlebot3 #788: Unable to roslaunch turtlebot3_bringup

PraveenElango, 2021-09-30:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [X ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [X ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [X ] etc (Jetson Nano)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ X] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ X] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    Run roscore from PC
    Run roslaunch turtlebot3_bringup turtlebot3_robot.launch on the Jetson Nano 

8. Copy and Paste the error messages on terminal.

    RLException: ERROR: unable to contact ROS master at [10.42.0.1:11311]

9. Please describe the issue in detail.

    The IP address of my PC is 10.42.0.1 and that of my Jetson Nano is 10.42.0.76. I installed the turtlebot3 packages on the Jetson and for the past 2 months I have been able to run roslaunch turtlebot3_bringup turtlebot3_robot.launch on the Jetson Nano without any issues. I always run roscore from my PC and my ROS_MASTER_URI and ROS_HOSTNAME are both set correctly on my PC and the Jetson in their respective .bashrc files as per: https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#pc-setup

However, I am now unable to call roslaunch turtlebot3_bringup turtlebot3_robot.launch and after a long 5 minutes, it just returns an error that says: RLException: ERROR: unable to contact ROS master at [10.42.0.1:11311]. I have tried pinging both my PC and my Jetson from each other using the aforementioned IP addresses and they can receive the signals without any issue. Therefore, I am unsure of where the problem is arising from. Would appreciate it if someone could point me in the right direction. Thanks!

ROBOTIS-Will, 2021-09-30:
Hi @PraveenElango 
ROS communicates to each node via a specific port.
I'd recommend to check if the port 11311 is not blocked on both Jetson Nano and the router.

PraveenElango, 2021-10-01:
Hi, @ROBOTIS-Will, thank you for your reply! It works now :)

closed 2021-10-01 by PraveenElango